lgsvl/simulator #292: 2019.07 version build from source error

Fibonacci43, 2019-08-21:
Hello, I follow [the instruction here](https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md) to build the simmulator from source, but I cannot get the same result when follow step 15. I check the unity project, but cannot find the scene named Borregasave.
My git clone process shows as below:
[image]
Step 15 shows as below(I only find one scene but names Loaderscene):
[image]

EricBoiseLGSVL, 2019-08-21:
@Fibonacci43 

You need to clone the Environments and Vehicles repos into the External folder. This is missing in our documentation and we are updating it soon.

[image]

martins-mozeiko, 2019-08-21:
@Fibonacci43  We do not put environments and vehicles in our main simulator repository to save size. Maps and vehicles can get pretty big in size because of 3d assets and texture.

We have example environments available on our github: [CubeTown](https://github.com/lgsvl/CubeTown) and [ SingleLaneRoad](https://github.com/lgsvl/SingleLaneRoad). And example vehicle [Jaguar2015XE](https://github.com/lgsvl/Jaguar2015XE). We'll be updating these with more eventually.

daohu527, 2019-08-22:
@martins-mozeiko @EricBoiseLGSVL Thanks for your help!
Some addition: 
1. It's not necessary to build maps and vehicles in simulator, because you can download the build maps and vehicles form the web. Just build the simulator is ok!
2. We still want the maps and vehicles models if we want to create a hd-map or a new car by ourselves. So I think lgsvl should provide both solutions.
3. Cloud you pls provide the models files in baidu pan. The download speed is too slow in github. Especially there is a 1.5G file!!!

EricBoiseLGSVL, 2019-08-22:
@daohu527  Thank you for helping others, you rock.
1) yep
2) yep that is why we are providing the repos and/or bundled assets. (when we can license wise)
3) we are looking into ways to distribute better, thanks for the info

luqiang21, 2019-08-23:
@daohu527 We have uploaded asset bundles and simulator binaries for version 2019.07 in Baidu pan. You can download from there.

Fibonacci43, 2019-08-24:
@EricBoiseLGSVL @martins-mozeiko  Thank you for your relay. Is there a plan to release the enviroment model and vehicle model? I want to make some **modification** to the environment and vehicle to feed my own need.

zelenkovsky, 2019-08-24:
Download new maps and vehicles from: [https://www.lgsvlsimulator.com/](https://www.lgsvlsimulator.com/)

[image]

martins-mozeiko, 2019-08-25:
@Fibonacci43 We have some sample maps and vehicles already published on github repository. You can get Jaguar vehicle in [Jaguar2015XE](https://github.com/lgsvl/Jaguar2015XE) repository. And for simple maps you can get [CubeTown](https://github.com/lgsvl/CubeTown) and [SingleLaneRoad](https://github.com/lgsvl/SingleLaneRoad). You can Use SingleLaneRoad Unity scene to build more complex road if you need one. We plan to put up some lager maps sometime later.

daohu527, 2019-08-26:
well done guys!

closed 2020-04-07 by EricBoiseLGSVL