lgsvl/simulator #293: how to open project in Unity on Ubuntu

asimonov, 2019-08-21:
hi, I follow updated instructions here https://github.com/lgsvl/simulator/blob/master/Docs/docs/build-instructions.md but I am stuck on this step:

* In the Projects tab, click Add and select the folder that the Simulator was cloned to

I select the 'simulator' folder but it does not appear on that page. I can only 'add' a new project.
I use the specified versions of UnityHub and Unity on Ubuntu18.
If I try to start Unity standalone to open the project I get UnityHub and i am back to square one.

alexiskovi, 2019-08-21:
Same issue here.
I can only create new empty project, but it also disappears after re-opening Unity Hub or pressing "Add" button. 
So, problem is, "Add" button deleting all current projects and giving empty list, no matter it is my new  project or simulator project.

[image]
[image]

daohu527, 2019-08-21:
I have follow the step and it works well, Here are some questions that are similar to yours, you can try it as below.
[here](https://answers.unity.com/questions/1342685/cant-create-a-new-project-in-unity-56.html)

asimonov, 2019-08-21:
that link did not solve my issue. i tried logging out/in to unityhub but there is no difference.

EricBoiseLGSVL, 2019-08-21:
@asimonov  and @alexiskovi 

Thanks, @daohu527 , hopefully that helps some people.  If it doesn't, be sure to select the project folder that directly contains Unity Assets folder, not an empty root folder that contains it.  Hope this helps.

asimonov, 2019-08-21:
@EricBoiseLGSVL selecting Assets folder does not help either.

i see there are docker files provided. may work better for me. are they working?

martins-mozeiko, 2019-08-21:
I think you misunderstood what Eric is saying. You need to select folder which is parent of `Assets` folder.

You can also try running unity directly and opening project without Hub. Launch Unity binary, select "Open Project" button and browse to simulator location.

We use Docker file to build simulator on Jenkins CI. You can try to use it, but it will be more difficult that simply opening simulator in Unity Editor.

daohu527, 2019-08-22:
Maybe you can take a look at the log file, 
```
Q: Where can I find the Unity Hub logs?
A: Logs are here:
Windows: `%UserProfile%\AppData\Roaming\UnityHub\logs`
Mac: `~/Library/Application support/UnityHub/logs`
Linux: `~/.config/UnityHub/logs`
```
You can try to open the project directly with Unity and view the Unity log too.
```
Q: Where can I find the Unity Hub logs?
A: Logs are here:
Windows: `%UserProfile%\AppData\Local\Unity\Editor\Editor.log`
Linux: `~/.config/unity3d/Editor.log`
```

martins-mozeiko, 2019-08-22:
Unity Editor has one more location on Linux: `~/.config/unity3d/Unity/Editor/Editor.log` Sometimes it is more useful.

alexiskovi, 2019-08-22:
Thanks, will try other options. 
To my mind, it's Unity Hub issue, because Hub is losing links to all projects after pressing "Add", even my empty created projects.

EricBoiseLGSVL, 2019-08-22:
Try to make a new project first, open it, then try to add Simulator project.

EricBoiseLGSVL, 2019-08-22:
Have you tried [this](https://forum.unity.com/threads/cant-add-existing-project-with-unity-hub-2-0-0-if-there-arent-currently-any-projects.688354/)?

shalinmehtalgsvl, 2019-08-22:
I followed our instructions here: https://www.lgsvlsimulator.com/docs/build-instructions/
and had the same issue at step 11. I created a new project (with the New button) and then was able to add the simulator project to Unity Hub.

alexiskovi, 2019-08-23:
@shalinmehtalgsvl, for my instance, "Add" button deletes all previous objects from list so I can't open any projects. I have created three projects, but after adding simulator (or any other project) Unity Hub losing all projects and shows "You have not projects here"
Will try at other PC.

martins-mozeiko, 2019-08-23:
Maybe try removing UnityHub folder at `~/.config/UnityHub` location, maybe it cannot correctly create files there.

asimonov, 2019-08-23:
I think i tried it. But will try all suggestions again when i get to my desktop

_____________________________
Kind Regards,
Alexey Simonov

EricBoiseLGSVL, 2019-10-09:
Anyone that has this issue, please update Unity Hub and this will be resolved

closed 2020-04-07 by EricBoiseLGSVL