lgsvl/simulator #295: Cannot Start Dreamview

yoyouwen, 2019-08-21:
I have downloaded the LGSVL Branch of Apollo 3.0 and build it. Then when I try bash scripts/bootstrap.sh, it gave me feedback:
 
[image]

And when I open browser for http://localhost:8888/, I get refuse to connect.

martins-mozeiko, 2019-08-21:
Can you please check dreamview log file fully - does it have any error messages at bottom of file?

yoyouwen, 2019-08-21:
Here is the bottom of file:
[image]
 I think there is no error messages.

yoyouwen, 2019-08-21:
> Can you please check dreamview log file fully - does it have any error messages at bottom of file?

Here is the bottom of file:
[image]

I think there is no error messages.

martins-mozeiko, 2019-08-22:
This error probably means that there is something else running on port 8888. Check what other processes you have listening on port 8888.

closed 2020-04-07 by EricBoiseLGSVL