lgsvl/simulator #296: Starting Autoware with LGSVL

Felixlxc, 2019-08-22:
Hi,

   I'm trying to make Autoware work with LGSVL. And I want to load in the my_localization.launch file. And the error in the figure occurs. 
[image]
I wonder if the Autoware has to be the version for LGVSL and use it in docker or is it the issue of GPU? I'm using Autoware v1.12 and build from source.

   Thanks for the help!

martins-mozeiko, 2019-08-22:
If you are using latest simulator version (2019.07) then it works with Autoware 1.12 official docker without any changes. We have documentation how to run it here: https://www.lgsvlsimulator.com/docs/autoware-instructions/

Error message you have says that this launch file does not exist. Check that it really exists. For Autoware configuration/installation issues I suggest asking questions on their gitlab/forum, as we are no longer maintaining our fork of Autoware - our simulator works with Autoware code without modifications.

Felixlxc, 2019-08-22:
@martins-mozeiko Thanks for the reply! I find that the simulator has been updated, does it matter if I'm still using an earlier version? 
   And thanks for noticing about gitlab.

martins-mozeiko, 2019-08-22:
We have not tried earlier simulator version with 1.12. It probably works, but I'm not 100% sure.
Earlier versions we were testing only with out own fork of earlier Autoware version.

Felixlxc, 2019-08-22:
@martins-mozeiko  Thanks for the information!

closed 2019-08-22 by Felixlxc