lgsvl/simulator #297: does 2019-07 version support multi lidar?

yzl96, 2019-08-22:


martins-mozeiko, 2019-08-22:
Yes

yzl96, 2019-08-22:
@martins-mozeiko , just add one more lidar in json is ok, and I have another question, I want to change the lidar's parameter, like lidar's different beam's angle, I will change LidarSensor.cs.
For example, I have two lidars, and one is 32 beams, another is 64 beams, 32 beam's lidar's  32 beam have different angles, I now can change LidarSensor.cs to set the beams angle. Then for 64 beam's lidar, I also want to set it's beams angle, so now what I should do, since the LidarSensor.cs is change for 32 beams lidar?

martins-mozeiko, 2019-08-22:
You do not need to change C# source file to modify just parameters of sensor.
All parameters are specified in json you put on vehicle.
Check example for Apollo - it has two "Color Camera" sensors with different parameters: https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/
You can do exactly same for two lidars - just have different values there.

yzl96, 2019-08-23:
@martins-mozeiko , I see in the C# source file, the lidar's angle between one beam and another beam is the same, my lidar's angle between one beam and another beam is different, so I change the C# file.

martins-mozeiko, 2019-08-23:
Oh, I see. You want to change lidar beam ray distribution - in such case yes, you'll need to change C# code. Currently Lidar supports only uniform distribution.

If you put public bool as member to sensor then it will be possible to set it from json. Later we'll expose these kind of parameter in WebUI as separate controls. Currently you'll need to add this value manually to json string. But in general this will work fine.

closed 2020-04-07 by EricBoiseLGSVL