lgsvl/simulator #298: how can I use old version's map?

yzl96, 2019-08-22:
I want to use old version's map: sanfransico, if it's possible? if possible, How can I do?

martins-mozeiko, 2019-08-22:
Not in new simulator. We now use Unity's HDRP rendering which requires changing material for all assets to support higher quality of rendering. Porting it to new simulator will require significant amount of work. Pull requests are welcomed :)

That said we will be providing more extra maps in new simulator.

EricBoiseLGSVL, 2019-08-22:
Since the new simulator is using HDRP you will need to do update the old scene to get it to work.

1) update all materials to the new HDRP materials including our shader graph material, EnvironmentSimulation, for roads, curbs, grass, and parking lots. All drive-able surfaces.

2) Remove any Directional lights from scene.  We now use a procedural sky with a spawned HDRP directional light.

3) Remove all cameras, we spawn the needed HDRP camera at runtime.

4) You should only have the map mesh, map annotation, map origin and spawninfo in scene

5) You will need to recreate the HDMap.  We have refactored and expanded the Annotation tool to support more features but it breaks the old format.

6) You need to use our build process to create an asset bundle to load in the Simulator webUI. Simulator -> Build...  You can test your scene in editor, all needed simulator objects will spawn in editor if you have updated correctly, just press play in scene.

I recommend using a part of San Francisco mesh that you need, not the entire map.

zelenkovsky, 2019-08-24:
@yzl96 You can get more maps from here: https://content.lgsvlsimulator.com/

We have a comprehensive plan to provide many different environments, allow users to create and share their own.

[image]

Francisobiagwu, 2019-09-27:
I believe that a tutorial youtube video and an update on the github documentation/readme will help. It may be easy for people already familiar with the tool to know exactly what to look for, but for starters, it is really difficult.

EricBoiseLGSVL, 2019-09-27:
@Francisobiagwu 

We completely agree.  We have many tutorial videos on our road map and will be releasing as soon as we can.

EricBoiseLGSVL, 2020-04-07:
@Francisobiagwu we will be releasing new tutorials and updated documentation soon

closed 2020-04-07 by EricBoiseLGSVL