lgsvl/simulator #299: can I change the reflection intensity for lidar?

yzl96, 2019-08-22:
Hi, I look the point cloud of simulator's lidar in rviz, the reflection intensity of point cloud is not very suitable for real world, so I want to change the reflection intensity, if I can change the reflection intensity , can you tell me the function I should change?

bigsheep2018, 2019-08-22:
same question here.
I noticed that there is no uses of Unity3D's built-in function like RayCast https://docs.unity3d.com/ScriptReference/Physics.Raycast.html within LidarSensor.cs

I am wondering where I could access the lidar ray's direction for customizing the lidar's shooting angle and, access those collision points'  material, normal vector, etc. for further customizing my intensity value, respectively.

I am totally new to C#, please specify the code with more details.
Thanks for releasing this wonderful work.

bigsheep2012, 2019-08-22:
@martins-mozeiko

martins-mozeiko, 2019-08-22:
We do not use RayCast for lidar simulation because it is very slow, because it runs on CPU. Instead we do lidar simulation on GPU by rendering depth to the texture and then reading it back on CPU to get distances. GPU executes many more instructions in parallel which is perfect for lidar - as it is shooting many rays, not one by one.

Actual calculation of intensity happens in GPU shader here: [LidarPass.hlsl](https://github.com/lgsvl/simulator/blob/master/Assets/Shaders/LidarPass.hlsl) This is part of fragment shader that is integrated into HDRP pipeline. LidarSensor.cs is only setting up input properties and invoking this shader.

Currently shader reads intensity from special texture that is assigned to the objects (line 49) - which allows us to set intensity value explicitly for every single pixel of surface. Alternatively it does simple grayscale color if this intensity texture does not exist (line 51).

Normal vector should be available here - either in surfaceData or builtinData.

closed 2019-10-02 by martins-mozeiko