lgsvl/simulator #300: Run my own created scene, the object is lost

liuppppp, 2019-08-22:
Hey, lg's team, first of all, thank you for developing such a great product.
I have a problem.
I imitated the scene of Sanfan City and made a scene myself, including camera, lighting, environment, map segment and so on. It shows normal in scene mode, and there is no problem in running alone.
I added this scene to the environment options of the menu. After running, the object objects in my scene disappeared.
Excuse me, where am I going wrong?
thank

martins-mozeiko, 2019-08-22:
Which objects exactly are disappearing? Are they not showing up in game view? Or do they not exist in scene hierarchy at all?

EricBoiseLGSVL, 2019-08-22:
@liuppppp Glad you like the new simulator.

Since the new simulator is using HDRP you will need to do a few things to your scene to get it to work.

1) update all materials to the new HDRP materials including our shader graph material, EnvironmentSimulation, for roads, curbs, grass, and parking lots. All drive-able surfaces.

2) Remove any Directional lights from scene.  We now use a procedural sky with a spawned HDRP directional light.

3) Remove all cameras, we spawn the needed HDRP camera at runtime.

4) You will need to recreate the HDMap.  We have refactored and expanded the Annotation tool to support more features but it breaks the old format.

5) You need to use our build process to create an asset bundle to load in the Simulator webUI. Simulator -> Build...  You can test your scene in editor, all needed simulator objects will spawn in editor if you have updated correctly, just press play in scene.

Like @martins-mozeiko said, more information on what is happening is needed.

liuppppp, 2019-08-23:
sorry, I hope this description will be clearer.
Road object disappeared. I have a highway in my city. I have HD map data for this road. I wrote it as an OSM file, then parsed the OSM file, generated a mesh in unity, and attached materials, textures, colliders, etc. 
So, this road exists in the hierarchy, I added an ego car, which can also appear in the game view. The problem is that I added this scene to the map's selectable menu, trying to get into the scene from the menu scene, in which my path disappeared.

1.in the scene view, seems that everything looks normal
[image]
2.Run this scene separately, in the game view, looks good too
[image]
3.but, added this scene in the menu list, my road object disappered
[image]
[image]

I hope you can understand what I am expressing and give me some advice. thx !

liuppppp, 2019-08-23:
My road has appeared
Thank you very much for your suggestion, very useful

zelenkovsky, 2019-08-24:
@ liuppppp Please try our latest version of Simulator.

Firstly we have greatly improved as workflow for creating new environments and maps.
1) New "Simulator" menu
2) Improved Map Annotation Tool [https://www.lgsvlsimulator.com/docs/map-annotation/#creating-a-new-map](https://www.lgsvlsimulator.com/docs/map-annotation/#creating-a-new-map)
3) HD Map import from Lanelets2 
4) Special __Simulator/Developer Settings...__ menu to spawn vehicle automatically during development.
5) __MOST IMPORTANT__: now you can build your own maps in to independent bundles to be distributed separately from Simulator: https://content.lgsvlsimulator.com/

BTW: if you have a good environment - we can publish that on our website.

closed 2019-10-02 by martins-mozeiko