lgsvl/simulator #301: Can't start 2019.07

vatna, 2019-08-23:
I've tried to run the Linux binary on Ubuntu 18.04 and I get the screen to select resolution & graphics quality but after clicking ok, I get nothing. It was complaining that I didn't have canberra-gtk-module, so have installed that. Now I don't get any error message. Is there anything else that I need to install?
[image]

shalinmehtalgsvl, 2019-08-23:
Hi @vatna ,

Do you have vulkan installed? These libraries are required on Linux.

`sudo apt install libvulkan1`

martins-mozeiko, 2019-08-23:
And if that does not help check the log file at `~/.config/unity3d/Editor/Player.log` location for any error messages.

canberra-gtk-module is not needed, this is some internal GTK dependency which is optional. We do not directly use it.

vatna, 2019-08-23:
Thanks @shalinmehtalgsvl and @martins-mozeiko. It works now that I have vulkan installed.

closed 2019-08-23 by vatna

Kailiangdong, 2019-09-20:
@vatna What version of nvidia driver you use? I install lastest driver and always black screen after that

vatna, 2019-09-20:
@Kailiangdong I used the latest 430.26. My issue was resolved by installing vulkan.

Kailiangdong, 2019-09-21:
Thank you. You install driver with  .run file or install from ppa?

vatna, 2019-09-23:
@Kailiangdong I use `sudo ubuntu-drivers autoinstall`

martins-mozeiko, 2019-09-23:
We usually install nvidia drivers on ubuntu from https://launchpad.net/~graphics-drivers/+archive/ubuntu/ppa

Kailiangdong, 2019-09-24:
@vatna @martins-mozeiko  Thank you. Yesterday, when I run lgsvl. The computer always crashes and frozen.  I search it is nvidia driver is not stable on linux.
Do you blacklist nouveau before installing nvidia-drivers？

martins-mozeiko, 2019-09-24:
I believe you don't need to do that. Installing binary nvidia driver will automatically blacklist nouveau driver by creating blacklist entry in `/etc/modprobe.d` folder.