lgsvl/simulator #302: Simulator crash with Apollo 5.0

vatna, 2019-08-23:
After starting bootstrap.sh and bridge.sh, I start up the simulator using the supplied "BorregasAve, noninteractive (with Apollo 5.0)". While in Module Controller, as I'm starting up the modules, the simulator will shut down. Sometimes it happens after I've started a few of the modules, but most of the time it happens after I start the first module (localization). Any suggestions as to what I should be doing differently?

martins-mozeiko, 2019-08-23:
Please check the log file at `~/.config/unity3d/Editor/Player.log` location for any errors at the bottom.

vatna, 2019-08-23:
Having looked at the log file and seeing "unloading 14 objects...", I tried reducing the resolution & graphics quality and it did work when using a fairly low resolution (960x540). So I'm guessing it's memory on the GPU? I have a GTX1060 6GB card.

I'm now able to start all the modules, and define a destination point. I see that the route to that point is then shown in the Dreamview window, however the car does not move.
[960_540_Player_nocrash.log](https://github.com/lgsvl/simulator/files/3536197/960_540_Player_nocrash.log)
[1024_640_Player_crash.log](https://github.com/lgsvl/simulator/files/3536198/1024_640_Player_crash.log)
[1280_1024_Player_crash.log](https://github.com/lgsvl/simulator/files/3536199/1280_1024_Player_crash.log)

martins-mozeiko, 2019-08-23:
Yes, this error is about running out of memory on GPU.

Our HDRP maps very high detail and it requires at least 8GB of GPU memory. If running with Apollo it might require even more - 10GB, because Apollo will use some memory for its perception module.

We have seen that Apollo 5.0 does not start moving car by its own. You need to "help" it a bit - press up arrow key for short time, after that car will move.

vatna, 2019-08-23:
Thanks @martins-mozeiko for confirming that it's a GPU memory issue. It's a bit upsetting though as 11GB cards seem to start at about $900. Anyhow, I'll just use a low res display for now.

As you suggested, giving the car a kick makes it move.

closed 2019-08-23 by vatna