lgsvl/simulator #304: Unity editor errors with 2019.07 version

riveranb, 2019-08-24:
I download source codes from 2019.07 release (https://github.com/lgsvl/simulator/releases/tag/2019.07)
and follow build instructions (https://www.lgsvlsimulator.com/docs/build-instructions/)

But when I open Unity project with Unity 2019.1.10f1 version, I get lots of errors, screenshots below
[image]

And I also followed step 22 to run "npm install" completely.
[image]

But still get those errors.
Errors shows some dependent namespace cannot be found
CS0246	The type or namespace name 'Google' could not be found (are you missing a using directive or an assembly reference?)
CS0246	The type or namespace name 'ProtoBuf' could not be found (are you missing a using directive or an assembly reference?)
CS0246	The type or namespace name 'FileDescriptorProto' could not be found (are you missing a using directive or an assembly reference?)

Is there something wrong? Or did I miss something to complete the build?

martins-mozeiko, 2019-08-25:
I think you do not have git-lfs installed. We use Git LFS file storage to make cloning a faster process. 
Make sure you have git lfs correctly installed before you clone repository: https://www.lgsvlsimulator.com/docs/faq/#why-are-assets-scenes-missing-empty-after-cloning-from-git

riveranb, 2019-08-25:
> I think you do not have git-lfs installed. We use Git LFS file storage to make cloning a faster process.
> Make sure you have git lfs correctly installed before you clone repository: https://www.lgsvlsimulator.com/docs/faq/#why-are-assets-scenes-missing-empty-after-cloning-from-git

Well, I did install git lfs already. Besides, I downloaded source code archive from release tag to build.
I also had tried cloning from git (LFS installed), all have the same errors.

martins-mozeiko, 2019-08-25:
Oh, I missed that you downloaded source from releases page. Unfortunately that will not work, because GitHub is not packaging LFS files inside .zip or .tar.gz archives when you download source code from releases page or tag, or any other place in commit history. 

Your only option is to do git clone if you want to work on source code. Check that git lfs has installed correct filters for your user. On Windows you should have `.gitconfig` file in your home folder (`%USERPROFILE%`) with following content:
```
[filter "lfs"]
	clean = git-lfs clean -- %f
	smudge = git-lfs smudge -- %f
	process = git-lfs filter-process
	required = true
```
If not, repeat installation process.

riveranb, 2019-08-25:
> [filter "lfs"]
> 	clean = git-lfs clean -- %f
> 	smudge = git-lfs smudge -- %f
> 	process = git-lfs filter-process
> 	required = true
> ```
> 
> If not, repeat installation process.

I retry with git clone, and succeed without errors. Thank you!
Last time I used GitKraken to clone, maybe my GitKraken didnot work with git lfs properly.

closed 2019-10-02 by martins-mozeiko