lgsvl/simulator #305: LoaderScene.unity

yumianhuli2, 2019-08-25:
1 When I enter LoaderScene.unity and press open browser, I get an empty web page.
2 My environment and vehicle folders are empty. How can I download these resources?Please tell me the reliable steps.
3 When can you update this repo and fix these bugs?
Thanks!

martins-mozeiko, 2019-08-25:
1. If you are running Loader scene from Unity Editor then you must do the build of WebUI (Simulator -> Build WebUI) Did it succeed without any errors?
If you are saying that you get same white page even when running simulator binary release - try refreshing browser cache, maybe it cached some old html files (press Ctrl+Shift+R after you open localhost:8080). If that does not help check the console output of browser (Ctrl+Shift+J on Chrome) for any error messages. Check also errors in log file of simulator binary at ~/.config/unity3d/Editor/Player.log` location.

2. You need to clone environments manually. We keep them in separate folders to save on space/bandwidth needed when cloning simulator. Not everybody will need all vehicles/environments. We provide few enviornments vehilces as open-source: [SingleLaneRoad](https://github.com/lgsvl/SingleLaneRoad/), [CubeTown](https://github.com/lgsvl/CubeTown) and [Jaguar2015XE](https://github.com/lgsvl/Jaguar2015XE).
Other vehicles and maps are available only as binary downloads. They get downloaded automatically when you first time run binary release of simulator. Or you can download asset packages manually and add them to WebUI. See following documentation links:
https://www.lgsvlsimulator.com/docs/maps-tab/#where-to-find-maps
https://www.lgsvlsimulator.com/docs/vehicles-tab/#where-to-find-vehicles

yumianhuli2, 2019-08-26:
>     1. If you are running Loader scene from Unity Editor then you must do the build of WebUI (Simulator -> Build WebUI) Did it succeed without any errors?
Hello,the Error is:
Error building WebUI
UnityEngine.Debug:LogError(Object)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:64)

martins-mozeiko, 2019-08-26:
Have you run `npm install` in WebUI folder? You need to have installed all required dependencies with npm.

For detailed instructions check our documentation about building from source: https://www.lgsvlsimulator.com/docs/build-instructions/

yumianhuli2, 2019-08-26:
> Have you run `npm install` in WebUI folder? You need to have installed all required dependencies with npm.
> 
> For detailed instructions check our documentation about building from source: https://www.lgsvlsimulator.com/docs/build-instructions/

Yes, I run npm install command in WebUI folder,And got the answer below:
~/XXX/apollo_5.0_simulator/WebUI [had (master)]$ npm install
npm WARN optional Skipping failed optional dependency /jest-haste-map/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.2.9
npm WARN optional Skipping failed optional dependency /chokidar/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.2.9

And then I Run Build WebUi... in Unity: Simulator -> Build WebUI...,got 
Error building WebUI
UnityEngine.Debug:LogError(Object)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:64

&_& PLs Help!

yumianhuli2, 2019-08-27:
Got it,the node version is so old.

yumianhuli2, 2019-08-27:
> Can you please run `npm run pack` in WebUI folder to see it will provide better error message?
Got it.the node version is so old.

but when I Open Build... in Unity: Simulator -> Build...
But the enviroments and vehicles are empty.And I also can not see BorregasAve.unity scene.
Just download SingleLaneRoad, CubeTown and Jaguar2015XE and put them into unity is enough for this question?But I even can not find something call BorregasAveXX.

srirambadri, 2019-09-19:
Hi @martins-mozeiko ,

I got a similar error as @yumianhuli2. I get ERROR BUILDING WEBUI.

I ran npm install and end up with the same error as  
[image]

martins-mozeiko, 2019-09-19:
BorregasAve scene is no open-source. We maintain this scene and are releasing only as binary bundle.

If you want to build some bundles manually, you need to place them in correct folders and then they will appear in build menu. Read more information in our documentation: https://www.lgsvlsimulator.com/docs/assets/

closed 2019-10-02 by martins-mozeiko