lgsvl/simulator #307: hdrp issue in ubuntu

yumianhuli2, 2019-08-25:
I found Render pipeline menu in window10, but not in ubuntu.

martins-mozeiko, 2019-08-25:
What do you mean by "Render pipeline menu" ?
And what do mean when saying "HDRP plug-in"? Do you mean Unity's HDRP package? It is in our github repository: [com.unity.render-pipelines.high-definition](https://github.com/lgsvl/simulator/tree/master/Packages/com.unity.render-pipelines.high-definition) We don't use upstream package as we needed to extend it to work better with our simulator.

yumianhuli2, 2019-08-26:
> What do you mean by "Render pipeline menu" ?
> And what do mean when saying "HDRP plug-in"? Do you mean Unity's HDRP package? It is in our github repository: [...]

Thanks! Then how can I use this com.unity.render-pipelines.high-definition  package?

martins-mozeiko, 2019-08-26:
I'm not sure I understand what are you asking. What do you mean by "use"? Unity uses this package automatically to render everything you see. You can read about HDRP rendering in Unity here: https://docs.unity3d.com/Packages/com.unity.render-pipelines.high-definition@5.6/manual/index.html

yumianhuli2, 2019-08-26:
> I'm not sure I understand what are you asking. What do you mean by "use"? Unity uses this package automatically to render everything you see. You can read about HDRP rendering in Unity here: https://docs.unity3d.com/Packages/com.unity.render-pipelines.high-definition@5.6/manual/index.html

Because I did find this package in ubuntu,but found it in WIN10

closed 2019-10-02 by martins-mozeiko