lgsvl/simulator #309: Missing vehicle and map in apollo-5 fork

man-csl, 2019-08-26:
I followed instructions in https://github.com/lgsvl/apollo-5.0 to setup the simulator and apollo-5 fork. However I dont see the XE_Rigged vehicle and San Francisco map loaded in the dreamview. In the repository I dont see these modules either. I've used the following command to clone the repo - 
git clone --recurse-submodules https://github.com/lgsvl/apollo-5.0.git

Could this be an issue from my side?

martins-mozeiko, 2019-08-26:
The instructions in readme seems to a bit outdated. Please follow instructions on our documentation page: https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/
We'll update the readme.

man-csl, 2019-08-26:
Thank you.

I cant see the mentioned vehicle (XE_Rigged_Apollo5.0) in dreamview still however.

[image]

Also, I don't see the mentioned vehicle nor map in the simulator but I am using 2019 05 version. Upgrading to latest should fix this I guess.

man-csl, 2019-08-26:
Just installed latest (2019 07) version of the simulator. I understand that it is cloud based now. It does not list the vehicle mentioned in the instructions - Lincoln2017MKZ (Apollo 5.0).
However it has following vehicle:
Lincoln2017MKZ (Apollo 5.0)

mmozeiko, 2019-08-26:
You're right, it's missing MKZ. We'll add it. For now select any other vehicle - everything should still work fine.

For previous version of simulator (<2019.07) you need to use `old-simulator` branch in Apollo repository: https://github.com/lgsvl/apollo-5.0/tree/old-simulator

martins-mozeiko, 2019-08-26:
You can select any other vehicle there - everything will still work fine.

For previous version of simulator (<2019.07) you need to use old-simulator branch in Apollo repository: https://github.com/lgsvl/apollo-5.0/tree/old-simulator

man-csl, 2019-08-26:
Got most of the things to work with latest simulator. The issue I face currently is that the vehicle does not move even after submitting a routing request. I have used the 'BorregasAve, evening (with Apollo 5.0)' simulation. 

Please see apollo screenshot below.

[image]

closed 2020-04-08 by EricBoiseLGSVL