lgsvl/simulator #311: About the Zoom Level

pengpingliang, 2019-08-28:
**Description:** 
Based on Simulator 2019.5, in order to make a new city map, I try to do them as following:

1. Create a city map FBX file with the MapBox package of Unity ,  and set Zoom Level 16.
(https://wiki.openstreetmap.org/wiki/Zoom_levels)
2. Load the FBX into the SanFrancisco scene.
3. The result is what the new city map's  is too small. 

and I guess the reason is what the map zoom level is not correct.

**Question:** 
Could you tell me how I can clear the problem ?  Thank you ~

pengpingliang, 2019-08-28:
@martins-mozeiko 
Thank you very much for you reply. I got it~
In order to make The Coordinate(X,Y,Z)  to match the UTM Coordinate , how to decide the proper scale value ? 
I guess the scale value depend to Zoom Level in the SanFrancisco Scene.

daohu527, 2019-08-28:
1. Adjust the model size by scale in Unity or other methods.
2. After the scale of the model is adjusted, you only need to set the origin and time zone of UTM in "MapOrigin". In this way, the coordinates in the game are related to the world coordinates. You need to know where the origin of the model is at the coordinates of the world.

pengpingliang, 2019-08-29:
@daohu527 
Thank you very much for your reply. The detail is very helpful to me.

what is "the model " in a map(FBX File)?   any road  or any building ?  We could not get the real size .

Now,  I try to adjust the scale of the map until any point's  UTM Coordinates  approach to the real value.
But it is difficult to get the very good result(precision)....

daohu527, 2019-08-29:
You know that the unit in Unity is meters. The model is the houses streets trees, etc. Or we change a word called city or environment. The size of the city just affects the local environment. Enlarging and shrinking the city does not change the origin of the city. If your city has an origin, this origin does not change, so I don't think there is any connection between the size and location of the city. You can take a screenshot of the model and we can see the coordinate.
You can also see the environment in "san francisco Scene". Hope this will helpful.

pengpingliang, 2019-08-29:
Thank you very much ~
I try to do it as your advice!

closed 2019-08-29 by pengpingliang

gaohongfein, 2019-09-18:
>     1. Adjust the model size by scale in Unity or other methods.
> 
>     2. After the scale of the model is adjusted, you only need to set the origin and time zone of UTM in "MapOrigin". In this way, the coordinates in the game are related to the world coordinates. You need to know where the origin of the model is at the coordinates of the world.
> 
> 
> ```
> Model (coordinates in the game) + MapOrigin = UTM (world real coordinates)
> ```
> 
> The coordinates in the game are usually (0,0,0), and you need to pay attention to the z-axis in the game corresponding is the y-axis of the world coordinates. The y-axis in the game corresponds to the z-axis (height) of the world.

can you tell me in which file and which line to change in order to match the world coordinate and game coordinate?

daohu527, 2019-09-18:
Which version do you use. you can find the "MapOrigin" in the "Hierarchy windows".