lgsvl/simulator #314: Why can't open the 2019.07 simulator release on Ubuntu?

OutOfCage, 2019-08-28:
Hi, LG Team.
I download lgsvlsimulator-linux64-2019.07 release from [Baidu Netdisk](https://pan.baidu.com/s/1M33ysJYZfi4vya41gmB0rw#list/path=%2F) and install Vulkan userspace library on Ubuntu(use command line `sudo apt-get install libvulkan1`). After I double-click the `Simulator` and then press `Ok`, nothing appears. 
This is the content of `/.config/unity3d/Editor/Player.log`.
```
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
Vulkan error VK_ERROR_INCOMPATIBLE_DRIVER (-9) file: ./Runtime/GfxDevice/vulkan/VKContext.cpp, line: 333
Vulkan error./Runtime/GfxDevice/vulkan/VKContext.cpp:333
Vulkan detection: 0
No supported renderers found, exiting
 
(Filename: ./PlatformDependent/LinuxStandalone/main.cpp Line: 639)
```
Thx.

david-gwa, 2019-08-29:
we had Nvidia Qudaro P2000 with [driver-390](https://www.nvidia.com/Download/driverResults.aspx/149145/en-us), looks work

david-gwa, 2019-08-29:
my local system works: 

Ubuntu 16.0.4
GPU: Qudaro P2000
Nvidia driver: 384

I didn't see a compatible vulkan driver for `nvidia-384`, but simple run `sudo apt-get install libvulkan1` works.

OutOfCage, 2019-08-29:
Thanks for help. I find a temporary way in issue #287. They have similar errors for vulkan. I use `sudo apt-get install mesa-vulkan-drivers vulkan-utils` to solve it because my gpu is Intel HD Graphics at present.

closed 2019-08-30 by OutOfCage