lgsvl/simulator #316: how to get assets content

liuppppp, 2019-08-29:
hi ,lg team
The new version you provided is really great.
I noticed that you have separated the scene and vehicle resources. They became resource files in the assetbundle format.
The scenes you provided are great. I was wondering how do I create a new file in assetbundle format. For example, the San Francisco scene you provided before, can I add it to the new version of the map list, what should I do?
Any suggestions, thank you

martins-mozeiko, 2019-08-29:
You can check our [SingleLaneRoad](https://github.com/lgsvl/SingleLaneRoad) or [CubeTown](https://github.com/lgsvl/CubeTown) repositories. These are examples what needs to be done to create map. After you create scene with contents you want it can be build with Simulator -> Build... menu item. This menu item will build AssetBundle from scenes in folders that are placed in Assets/External/Environments folder. So, for example, for CubeTown the build menu will look for `Assets/External/Environments/CubeTown/CubeTown.unity` scene file. You can place as many as you want folders inside this Environments folder.

liuppppp, 2019-08-30:
ok. thx

closed 2019-10-02 by martins-mozeiko