lgsvl/simulator #317: Why does NPC car's speed change all the time

JIXIMD, 2019-08-29:
Hi, LG Team
  I did a test, the script is as follows
[image]

  
[image]

  Why does NPC car speed keep changing over time?And the scope of the change is unclear.How do I get the exact speed of NPC car?
Thank you for your help

shalinmehtalgsvl, 2019-08-29:
NPCs currently follow simple Physics where the rigidbody is moved to a new position based on the "currentSpeed" (see https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Controllers/NPCController.cs#L594). 

The speed returned by the API is the difference in the NPC's rigidbody position divided by the FixedDeltaTime (https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Controllers/NPCController.cs#L783). This is a rough estimation of the speed. 

We'll look into a more accurate way of reporting the speed.

ntutangyun, 2020-02-06:
Hi, seems that the 3D detection ground truth sensor is not producing velocity and acceleration information (empty), as shwon in `cyber_monitor` by Apollo 5.0. 

I'd like to ask if the team is currently solving the speed report ? thank you.

EricBoiseLGSVL, 2020-02-06:
@ntutangyun we will look into this

shalinmehtalgsvl, 2020-02-07:
@ntutangyun 

I was able to reproduce your issue when the NPC had a 0 velocity. Are you getting empty information for NPCs with non-zero velocity?

ntutangyun, 2020-02-07:
@shalinmehtalgsvl Hi, thanks for the reply.  I used Python API to set the velocity of the NPC vehicle. 
all velocity information are empty.

below is the script i use.  Thank You.

```python
#!/usr/bin/env python3

import os
import lgsvl
import math

from signal import signal, SIGINT
from sys import exit

sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "127.0.0.1"), 8181)

def exit_handler(signal_received, frame):
  exit(0)

def init():
  if sim.current_scene == "BorregasAve":
    sim.reset()
  else:
    sim.load("BorregasAve")
  
  spawns = sim.get_spawn()
  ego_state = lgsvl.AgentState()
  ego_state.transform = spawns[0]
  
  ego_state.rotation.y = ego_state.rotation.y - 90
  ego_forward = lgsvl.utils.transform_to_forward(ego_state.transform)
  ego_right = lgsvl.utils.transform_to_right(ego_state.transform)
  ego_state.transform.position = ego_state.transform.position + (50 * ego_right)
    
  ego = sim.add_agent("Licoln_Ap5.0_obs_no_lidar_no_cam", lgsvl.AgentType.EGO, ego_state)
#   ego = sim.add_agent("Lincoln_Apollo_5.0_obstacles", lgsvl.AgentType.EGO, ego_state)
  ego.connect_bridge("localhost", 9090)

  # suv
  suv_state = lgsvl.AgentState()
  suv_state.transform.position = ego_state.transform.position + (ego_right * 4) - (ego_forward * 10)
  suv_state.transform.rotation = ego_state.transform.rotation
  suv_state.velocity = lgsvl.utils.transform_to_forward(suv_state.transform) * 3
  sim.add_agent("SUV", lgsvl.AgentType.NPC, suv_state)

if __name__ == "__main__":  
  signal(SIGINT, exit_handler)
  
  while True:
    init()
    sim.run(time_limit = 10.0)
```

EricBoiseLGSVL, 2020-04-07:
@daviduhm This 3d ground truth sending 0.0 velocity is fixed correct?

closed 2020-04-07 by EricBoiseLGSVL