lgsvl/simulator #319: can not open browser in linux

gaohongfein, 2019-08-29:
why i can't open browser with a no root user. I can open browser only with root, but it presents also Running Firefox as root in a regular user's session is not supported.  ($HOME is /home/drl which is owned by drl.)

gaohongfein, 2019-08-29:
Logging to /home/drl/.config/unity3d/Editor/Player.log
Running Firefox as root in a regular user's session is not supported.  ($HOME is /home/drl which is owned by drl.)
Running Firefox as root in a regular user's session is not supported.  ($HOME is /home/drl which is owned by drl.)
Running Firefox as root in a regular user's session is not supported.  ($HOME is /home/drl which is owned by drl.)
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: iceweasel: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: seamonkey: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: mozilla: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: epiphany: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: konqueror: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: chromium-browser: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: google-chrome: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: www-browser: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: links2: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: elinks: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: links: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: lynx: not found
/usr/bin/xdg-open: 778: /usr/bin/xdg-open: w3m: not found
xdg-open: no method available for opening 'http://localhost:8080/'

martins-mozeiko, 2019-08-29:
What Linux distribution are you using? We officially support only Ubuntu 16.04 and up.

I think this is firefox configuration issue. If you have run fiefox as root it may have created files with "root" owner in your home folder. You need to make sure permissions of files and folders in your home folder are correct.

Check the answers here: https://askubuntu.com/questions/1037052/running-firefox-as-root-in-a-regular-users-session-is-not-supported-xauthori

closed 2020-04-07 by EricBoiseLGSVL