lgsvl/simulator #321: Invalid Simulation when running with "ROS" bridge

sriharshakunda, 2019-08-30:
I am trying to run the simulator with ROS bridge. I followed all the instructions from documentation but when I try to click the play button in simulation tab of WEBUI, an Invalid status shows up. I ran  roslaunch rosbridge_server rosbridge_websocket.launch  command in the terminal before launching the simulation.

martins-mozeiko, 2019-08-30:
Common issue is that connection string needs to be in `HOST:PORT` format. For example, `localhost:9090`. Check that you are using this format.

If that does not help, please open log file to see what kind of error messages are there.
On Linux it is in `~/.config/unity3d/Editor/Player.log` location.
On Windows it is in `%APPDATA%\..\LocalLow\LG Silicon Valley Lab\LGSVL Simulator\Player.log` location.

sriharshakunda, 2019-08-30:
@martins-mozeiko This is my output from roslaunch command

registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
[INFO] [1567125607.789471]: Rosbridge WebSocket server started on port 9090

WebUI is running on port 8080 and host is localhost

sriharshakunda, 2019-08-30:
@martins-mozeiko  below is my player log output. It seems like the host and port name a wrong? Can you tell me how to change that?

> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> Listing simulations
>  
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> Starting simulation with id 1
>  
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> Unloading 1 Unused Serialized files (Serialized files now loaded: 2)
> WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'IndirectDXR' has no vertex shader
> WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'ForwardDXR' has no vertex shader
> WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'VisibilityDXR' has no vertex shader
> WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
> UnloadTime: 0.599843 ms
> WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'IndirectDXR' has no vertex shader
> WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'ForwardDXR' has no vertex shader
> WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'VisibilityDXR' has no vertex shader
> WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
> Failed to start 'case1' simulation
>  
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> Exception: Incorrect bridge connection string, expected HOSTNAME:PORT
>   at AgentManager.SpawnAgent (Simulator.AgentConfig config) [0x00083] in <850181ed9fa248febe8c381050f5b5fc>:0 
>   at AgentManager.CreateAgentsFromConfigs (Simulator.AgentConfig[] agentConfigs) [0x0010e] in <850181ed9fa248febe8c381050f5b5fc>:0 
>   at AgentManager.SpawnAgents (Simulator.AgentConfig[] agentConfigs) [0x00000] in <850181ed9fa248febe8c381050f5b5fc>:0 
>   at SimulatorManager.Init (System.Nullable`1[T] seed) [0x00281] in <850181ed9fa248febe8c381050f5b5fc>:0 
>   at Simulator.Loader.SetupScene (Simulator.Database.SimulationModel simulation) [0x001d0] in <850181ed9fa248febe8c381050f5b5fc>:0 
> UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
> UnityEngine.DebugLogHandler:LogException(Exception, Object)
> UnityEngine.Logger:LogException(Exception, Object)
> UnityEngine.Debug:LogException(Exception)
> Simulator.Loader:SetupScene(SimulationModel)
> Simulator.<>c__DisplayClass27_1:<StartAsync>b__1(AsyncOperation)
> UnityEngine.AsyncOperation:InvokeCompletionEvent()
>  
> (Filename: <850181ed9fa248febe8c381050f5b5fc> Line: 0)
> 
> Build Timestamp = 8/23/2019 12:52:54 PM
>  
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> Version = 
>  
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> GitCommit = 
>  
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> GitBranch = 
>  
> (Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)
> 
> NullReferenceException: Object reference not set to an instance of an object
>   at VehicleController.UpdateLights () [0x0001b] in <850181ed9fa248febe8c381050f5b5fc>:0 
>   at VehicleController.Update () [0x00006] in <850181ed9fa248febe8c381050f5b5fc>:0 
>  
> (Filename: <850181ed9fa248febe8c381050f5b5fc> Line: 0)
> 
> 
> Unloading 211 unused Assets to reduce memory usage. Loaded Objects now: 64000.

martins-mozeiko, 2019-08-30:
You set connection when you are adding vehicle to simulation: https://www.lgsvlsimulator.com/docs/simulations-tab/

In this window: https://www.lgsvlsimulator.com/docs/images/full_size_images/web-simulation-mapVehicle.png

You should put there `localhost:9090` Change `localhost` to different hostname in case you are running bridge on different machine.

sriharshakunda, 2019-08-30:
@martins-mozeiko Thank you very much, I just realized that.

closed 2019-09-03 by sriharshakunda