lgsvl/simulator #323: keyboard can no control the vehicle in version 2019.07

askariz, 2019-08-30:
I refer the instruction from [](https://www.lgsvlsimulator.com/docs/) and I can't control the vehicle throught keyboard and can't find the Api both in the source code and release zip file ?

daohu527, 2019-08-30:
There is a doc about Keyboard shortcuts, [Here](https://www.lgsvlsimulator.com/docs/keyboard-shortcuts/).
Also API is support. [Python API](https://www.lgsvlsimulator.com/docs/python-api/)

askariz, 2019-08-30:
@daohu527 
yes I following the keyboard shortcuts and ok except vehicle control.

daohu527, 2019-08-30:
What do you mean the vehicle control?  You can control the car by
```
← ↑ ↓ → - Drive vehicle forward/brake, turn
```
I have tried this is ok.

askariz, 2019-08-30:
@daohu527 
OK. I Have found out the problem is that I don't add the sensors json format of the vehicle . after that I can control the vehicle now . thx!

closed 2019-08-30 by askariz

Francisobiagwu, 2019-08-30:
Please where is the python API located? After the new update, the documentation/instruction have some outdated information. It will be nice if someone can follow the instruction provided on the website/github and try to reproduce to fix the errors.
Thanks