lgsvl/simulator #324: AutonomouStuff Parking Lot can't work in Apollo?

daohu527, 2019-08-30:
I try the "Borregas Avenue" and "Single Lane Road" map, They work well in dreamview and I can see the road. But the **"AutonomouStuff Parking"** seems not work. Does anyone else have the same mistake?

```
Env: Window 10
Unity: 2019.1.10f1
Apollo3.5 Map:  [Map](https://github.com/lgsvl/apollo-3.5/tree/simulator/modules/map/data)
```
I download the hdmap for lgsvl/apollo-3.5 and download the unity assets from baidu pan. I am not sure where is the problem, so come up and ask the following.

martins-mozeiko, 2019-08-30:
What do you mean by "not work" ? Does it not localize vehicle? Or the hd map does not show up in dreamview? Or routing doesn't start? Or vehicle doesn't start driving?

daohu527, 2019-09-01:
1. I can see the car but hd map does not show up in dreamview

martins-mozeiko, 2019-09-05:
I verified that 3.5 with AutonomouStuff map should work fine.
Have you enabled all necessary modules in dreamview?
Can you check with cyber_monitor in terminal what topics are being sent from simulator?

daohu527, 2019-09-06:
Thanks for you reply !  

I will check again what's the problem, I did the same on other maps, but only this map has problems. I am downloading the map [here](https://github.com/lgsvl/apollo-3.5/tree/simulator/modules/map/data).

And I can see the car seems that the car's position was sent, anyway I will check again.

daohu527, 2019-09-06:
When I select the "AutonomouStuff Parking" map, I will see an error in Chrome. The error below seems to be some protobuf compatibility issues. I check the proto file with baidu apollo, and there is no difference with **"map.Id"**. I will see if there are other problems.
```
Uncaught Error: invalid wire type 4 at offset 2027
    at p.skipType (worker.bundle.js:1)
    at m.Id$decode [as decode] (eval at n (worker.bundle.js:1), <anonymous>:14:5)
    at m.Lane$decode [as decode] (eval at n (worker.bundle.js:1), <anonymous>:11:17)
    at m.Map$decode [as decode] (eval at n (worker.bundle.js:1), <anonymous>:26:24)
    at worker.bundle.js:1
```

closed 2020-04-08 by EricBoiseLGSVL