lgsvl/simulator #325: missing pb_msgs in ros_bridge canbus top in version 2019.07

askariz, 2019-08-30:
after enable canbus vehicle config 
` {
    "type": "CAN-Bus",
    "name": "CAN Bus",
    "params": {
      "Frequency": 10,
      "Topic": "/canbus"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
},`
on the ros_bridge side get missing 
`[ERROR] [1567149251.432392]: [Client 1] advertise: Unable to load the manifest for package pb_msgs. Caused by: pb_msgs`
I know that pb_msgs is apollo define msg which come with proto . and I can just install with 'ros-melodic-pb_msgs' in this case . how can I use the canbus message ?
@daohu527

daohu527, 2019-08-30:
Here is a sample about the apollo 3.0 sensor params. [here](https://www.lgsvlsimulator.com/docs/apollo-json-example/)  
1. The TOPIC you set seems error, You can try and give more feedback.
2. By the way, you can test with [lgsvl/apollo3.0](https://github.com/lgsvl/apollo-3.0) before you meet your own needs.

askariz, 2019-08-30:
hi @daohu527 
I have tested with apollo 3.0 . and can recevice the canbus proto message . 
but for my ros which are aboth with ubuntu 16.04 ros kinect and ubuntu 18.04 melodic when I start rosbridge get the same error which is `[ERROR] [1567175346.220001]: [Client 0] advertise: data_class [<class 'modules.canbus.proto.chassis_pb2.Chassis'>] is not a class`　Ｉhave no idea of it .

martins-mozeiko, 2019-08-30:
I'm not sure what is ros-melodic-pb_msgs, but if it installs pb_msgs package correctly you still need patched rosbridge that supports pb_msgs. Make sure you are using this one: https://github.com/lgsvl/rosbridge_suite instead of upstream one.

Are you sure ros-melodic-pb_msgs installs all necessary Apollo messages?

askariz, 2019-08-31:
@martins-mozeiko there is no package such ros-melodic-pb_msgs . the steps I take are

1. copy py_proto/modules from apollo build gennerated to /opt/ros/melodic/lib/python2.7/dist-packages 
2. pb_msgs from [apollo-platform](https://github.com/ApolloAuto/apollo-platform/tree/master/ros/pb_msgs) then put it into catkin_ws then catkin_make . 

but the result is the same 
`[ERROR] [1567175346.220001]: [Client 0] advertise: data_class [<class 'modules.canbus.proto.chassis_pb2.Chassis'>] is not a class`

btw . I just want canbus message from lgsvl simulator . but currently version it seems output message is base on apollo pb_msgs format . I think if it is output more common msg format would be better!

askariz, 2019-08-31:
ok. I think I have misunderstanding . What I need is the car position and velocity . I can get it from gps sensors and imu and gps Odometry that is good for me enough . 
@martins-mozeiko 
thanks for you kindly help.

closed 2019-10-02 by martins-mozeiko