lgsvl/simulator #328: Failed to load module "Canberra-gtk-module"

Francisobiagwu, 2019-08-30:
Each time I try to open the BorregasAve, no bridge, data collection. I get this error. 
Failed to load module "canberra-gtk-module"
Aborted (core dumped)

[image]

martins-mozeiko, 2019-08-30:
This error message about canberra is harmless, it comes from some internal Unity dependencies which we do not use.

To understand what crashes here, check the log file at `~/.config/unity3d/Editor/Player.log` - there should be some error messages at the bottom.

One potential reason is that you do not have vulkan client libraries installed. Unity HDRP rendering on linux uses Vulkan instead of OpenGL. Check if you have `libvulkan1` package installed.

zhouchengming1, 2019-09-11:
Try this: sudo apt install libcanberra-gtk-module libcanberra-gtk3-module

martins-mozeiko, 2019-09-11:
@zhouchengming1  libcanberra is completely irrelevant here. It is some optional GTK dependency that we do not use. GTK libraries probably try to load it this module it exists. But everything still works if it does not exist.

closed 2019-10-02 by martins-mozeiko

ntutangyun, 2019-12-12:
> This error message about canberra is harmless, it comes from some internal Unity dependencies which we do not use.
>
> To understand what crashes here, check the log file at [...]

this actually solved the issue.

ashish-roopan, 2022-01-10:
@martins-mozeiko . I am getting the same error when running pose detection on GPU. The ```libvulkan1``` package is installed and can't find any path ```~/.config/unity3d/Editor/Player.log``` for checking the log file. Can you assist me on what to do ?