lgsvl/simulator #330: Where is the url to download the map

yumianhuli2, 2019-08-31:
Hello!
    Where is the url to download the map? If I download a map locally, how can I reference it? Can you give me an example?Thanks!

martins-mozeiko, 2019-08-31:
If you launch simulator binary first time then it will download maps and vehicles automatically. If you want to add maps to when you launch simulator in Editor then you need to get map URL from https://content.lgsvlsimulator.com You do **not** need to download it. Just copy & paste URL when you are adding map. It will download it automatically.

shalinmehtalgsvl, 2019-09-04:
If you do download it locally (or make your own map) you just need to enter the file path into the URL box.

e.g. `C:\Users\XXX\Desktop\environment_mars`

closed 2019-10-02 by martins-mozeiko