lgsvl/simulator #331: rosbridge

SeongHyunBae, 2019-08-31:
I execute "roslaunch rosbridge_server rosbridge_websocket.launch" on terminal.

But, rostopic list

/client_count
/connected_clients
/rosout
/rosout_agg

How can I get sensor msgs????

Please help me

SeongHyunBae, 2019-08-31:
[INFO] [1567268491.573642]: Client connected.  1 clients total.
[INFO] [1567268491.575438]: Client disconnected. 0 clients total.
[INFO] [1567268491.582334]: Client connected.  1 clients total.
[INFO] [1567268491.584277]: Client disconnected. 0 clients total.
[INFO] [1567268491.601169]: Client connected.  1 clients total.
[INFO] [1567268491.603581]: Client disconnected. 0 clients total.
[INFO] [1567268491.606162]: Client connected.  1 clients total.
[INFO] [1567268491.608288]: Client disconnected. 0 clients total.
[INFO] [1567268491.612239]: Client connected.  1 clients total.

connected and disconnected are repeated continuously

martins-mozeiko, 2019-09-01:
What rosbridge are you running? Is it upstream version or from our fork?
What version of simulator are you connecting to it? What vehicle are you connecting? Does it have correct bridge type set?

askariz, 2019-09-02:
@SeongHyunBae first you need to correctly setup sensors parameters . please refer [Sensor JSON](https://www.lgsvlsimulator.com/docs/sensor-json-options/)

SeongHyunBae, 2019-09-02:
It is solved thanks.

closed 2019-09-02 by SeongHyunBae