lgsvl/simulator #333: About importing HD maps now

yumianhuli2, 2019-09-01:
Is there  support for importing maps and generating scenarios in Apollo or other formats now?

martins-mozeiko, 2019-09-01:
Latest simulator version supports importing HD map from Lanelet2 format. We are working on supporting also import from OpenDrive format. This will import only road annotations. Creating 3d geometry is still a manual process you'll need to do.

Scenarios can be created with our [PythonAPI](https://github.com/lgsvl/PythonAPI) - it supports all necessary functionality to set up specific initial cases (npcs, positions, waypoints, etc) and then check conditions you need - for example, reached position, collision events.

EricBoiseLGSVL, 2020-04-07:
@yumianhuli2 OpenDrive is now supported as well

closed 2020-04-07 by EricBoiseLGSVL