lgsvl/simulator #335: about front wheel angle

askariz, 2019-09-02:
in the new version of 2019.07 I can see the front wheel rotate when I turn left or turn right . how can I get the angle from which message ?

askariz, 2019-09-02:
@martins-mozeiko  currently I find canbus is output rosbridge with apollo pb_msgs but it is not standard msg with ros . I think it would better to make a more common msg output or move this into lgsvl_msgs maybe better .

martins-mozeiko, 2019-09-03:
The rationale for publishing pb_msgs is taht our simulator by default provides integration with Apollo or Autoware - thus it uses their messages. Canbus message was implemented because Apollo needed it. If you need this message for other AD stack, then it requires knowing what type to publish. We don't want to publish everything on custom messages from lgsvl_msgs package, as it would require a lot of effor and changes from AD stack developer in their code.

You can fairly easy add new ROS types that sensor will publish. Just check code for any existing sensor to see what needs to be done.

closed 2020-04-07 by EricBoiseLGSVL