lgsvl/simulator #337: why is the velocity of 3d_ground_truth always zero

JIXIMD, 2019-09-02:
Hi, lg team
[image]
[image]
   Here is my scenario and script,I want to get the npc car's speed,so I input command: "ros echo /simulator/ground_truth/3d_detections"(lidar was enabled) . The output is as follows, why is the velocity always zero.
[image]

thank you for your help

JIXIMD, 2019-09-04:
Does anyone have a similar problem?Or whose velocity is a truth value

daviduhm, 2019-09-05:
Hi @JIXIMD,
Thanks for your interests in our simulator and reporting this issue. We have verified the issue and fixed it in our internal repo. It will be pushed by the end of this week so please stay tuned. Having said that, we have updated to a new version of simulator (with bunch of new features and performance increased) and this fix will only be available in master branch in this repo. I can see from the screenshot that you are using the old version of our simulator. Let me kindly suggest you to migrate to the latest simulator since we will no longer support the old version.

JIXIMD, 2019-09-05:
Hi @daviduhm 
 Thanks for your reply.Yeah, I have tried the latest  version of simulator and encountered a new problem.
[image]
 I have inputed "git clone https://github.com/lgsvl/lgsvl_msgs.git" on my computer,But it doesn't fit the latest version of the simulator.Is there the latest version of lgsvl_msg?

daviduhm, 2019-09-05:
Your lgsvl_msgs package should be okay. This issue is because of a bug in the simulator code. It is a known issue and has already been fixed as well. Again, we are preparing a weekly release by the end of this week and all those issues regarding ground truth sensors will be resolved by that. Sorry about the inconvenience!

daviduhm, 2019-09-07:
@JIXIMD Let me ping you here to let you know that we have released [2019.09-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.09-rc1). The issues regarding ground truth sensors addressed in this thread have been fixed in the latest master branch!

JIXIMD, 2019-09-07:
I will try it, thank you very much!

JIXIMD, 2019-09-07:
@daviduhm, I tried the 2019.09-rc1 version and still had some problems.
1.The script is stuck in sim.load()
[image]
[image]
2.NPC vehicles are still, although I have set up my DriveWaypoint.

JIXIMD, 2019-09-09:
@daviduhm,I urgently need to use this simulator now. Could you please help me to look at the above mentioned problems?Is there something wrong with my operation?

daviduhm, 2019-09-09:
I'm not able to reproduce this issue. Could you check your log file and check if there are any exceptions? Assuming you're using the release binary on Windows, you can find a log file in `"%APPDATA%\..\LocalLow\LG Silicon Valley Lab\LGSVL Simulator\Player.log"`

JIXIMD, 2019-09-10:
**Player.log**

Mono path[0] = 'E:/lgsvl/lgsvlsimulator-windows64-2019.09-rc1/simulator_Data/Managed'
Mono config path = 'E:/lgsvl/lgsvlsimulator-windows64-2019.09-rc1/MonoBleedingEdge/etc'
Loading player data from E:/lgsvl/lgsvlsimulator-windows64-2019.09-rc1/simulator_Data/data.unity3d
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: NVIDIA GeForce GTX 1060 with Max-Q Design (ID=0x1c20)
    Vendor:   
    VRAM:     6052 MB
    Driver:   24.21.13.9836
Begin MonoManager ReloadAssembly
- Completed reload, in  1.292 seconds
WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - All passes removed
ERROR: Shader Shader is not supported on this GPU (none of subshaders/fallbacks are suitable)WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - Setting to default shader.
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
<RI> Initializing input.

New input system (experimental) initialized
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.

<RI> Initialized touch support.

WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
UnloadTime: 0.489200 ms
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - All passes removed
Current Database Version: 1
 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 48)

Final Database Version: 1
 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 48)

Setting up 6 worker threads for Enlighten.
  Thread -> id: 274c -> priority: 1 
  Thread -> id: 2be0 -> priority: 1 
  Thread -> id: 1f54 -> priority: 1 
  Thread -> id: 2c98 -> priority: 1 
  Thread -> id: 13dc -> priority: 1 
  Thread -> id: 898 -> priority: 1 
Fallback handler could not load library E:/lgsvl/lgsvlsimulator-windows64-2019.09-rc1/simulator_Data/Mono/libc
Fallback handler could not load library E:/lgsvl/lgsvlsimulator-windows64-2019.09-rc1/simulator_Data/Mono/libc.dll
Fallback handler could not load library E:/lgsvl/lgsvlsimulator-windows64-2019.09-rc1/simulator_Data/Mono/libc
Cannot create required material because shader is null
 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 48)

Cannot create required material because shader is null
 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 48)

Getting current user
 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 48)

Listing simulations
 
[... 43 lines cut ...]

Stopping simulation with id 1
 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 48)

Unloading 2 Unused Serialized files (Serialized files now loaded: 0)
<color=red>SimulatorManager Not Found!</color>
 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 48)

NullReferenceException: Object reference not set to an instance of an object
  at NPCController.ResetData () [0x00024] in <0b991c6960014f6e9eac95ee32de3df3>:0 
  at NPCController.OnDisable () [0x00000] in <0b991c6960014f6e9eac95ee32de3df3>:0 
 
(Filename: <0b991c6960014f6e9eac95ee32de3df3> Line: 0)

UnloadTime: 10.157700 ms

Unloading 46 unused Assets to reduce memory usage. Loaded Objects now: 4932.
Total: 7.198900 ms (FindLiveObjects: 0.307600 ms CreateObjectMapping: 0.146600 ms MarkObjects: 6.212900 ms  DeleteObjects: 0.531500 ms)

Unloading 0 Unused Serialized files (Serialized files now loaded: 0)

Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 4932.
Total: 6.923100 ms (FindLiveObjects: 0.574100 ms CreateObjectMapping: 0.203400 ms MarkObjects: 6.136200 ms  DeleteObjects: 0.009100 ms)

daviduhm, 2019-09-11:
Are you using a modified version of map? If you've modified a map, could you first try with the original map like Single lane road or Borregas ave and see if same error happens? For your modified map, how did you create this map and have you placed any new objects in a scene?

JIXIMD, 2019-09-11:
The above  Player.log was created using the Single lane road original map，that I download from https://content.lgsvlsimulator.com/maps/singlelaneroad/ yesterday。

daviduhm, 2019-09-12:
Ah I see. I think we've also verified and were able to reproduce this issue today. We are currently working on fixing this issue and will release another version in few days by the end of this week. I'm sorry about that.

In the meantime, you can try removing this line 
[SimulatorManager.Instance.AgentManager.ClearActiveAgents();](https://github.com/lgsvl/simulator/blob/14678160ee44e7026be5297a728652df39b81500/Assets/Scripts/Api/ApiManager.cs#L101) and moving it to [here](https://github.com/lgsvl/simulator/blob/14678160ee44e7026be5297a728652df39b81500/Assets/Scripts/Api/ApiManager.cs#L237). I think this could possibly fix your issue for now.

JIXIMD, 2019-09-12:
I have done what you said and recompiled with unity.But it didn't work,the player.log as follow:
Mono path[0] = 'D:/vmshared_space/lgsvl/map/simulator_Data/Managed'
Mono config path = 'D:/vmshared_space/lgsvl/map/MonoBleedingEdge/etc'
PlayerConnection initialized from D:/vmshared_space/lgsvl/map/simulator_Data (debug = 0)
PlayerConnection initialized network socket : 0.0.0.0 55165
Multi-casting "[IP] 192.168.10.72 [Port] 55165 [Flags] 2 [Guid] 2501081925 [EditorId] 1736037784 [Version] 1048832 [Id] WindowsPlayer(DESKTOP-3FBNM21) [Debug] 0 [PackageName] WindowsPlayer" to [225.0.0.222:54997]...
Started listening to [0.0.0.0:55165]
PlayerConnection already initialized - listening to [0.0.0.0:55165]
Loading player data from D:/vmshared_space/lgsvl/map/simulator_Data/data.unity3d
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: NVIDIA GeForce GTX 1060 with Max-Q Design (ID=0x1c20)
    Vendor:   
    VRAM:     6052 MB
    Driver:   24.21.13.9836
Begin MonoManager ReloadAssembly
- Completed reload, in  0.086 seconds
WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - All passes removed
ERROR: Shader Shader is not supported on this GPU (none of subshaders/fallbacks are suitable)WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - Setting to default shader.
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
<RI> Initializing input.

New input system (experimental) initialized
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.

<RI> Initialized touch support.

WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
UnloadTime: 1.251000 ms
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - All passes removed
Current Database Version: 1
UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
Simulator.Database.DatabaseManager:CheckForPatches() (at D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Database\DatabaseManager.cs:130)
Simulator.Database.DatabaseManager:Init() (at D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Database\DatabaseManager.cs:91)
Simulator.Loader:Start() (at D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Web\Loader.cs:120)
 
(Filename: D:/vmshared_space/lgsvl/simulator/Assets/Scripts/Database/DatabaseManager.cs Line: 130)

Final Database Version: 1
UnityEngine.DebugLogHandler:Internal_Log(LogType, LogOption, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
Simulator.Database.DatabaseManager:CheckForPatches() (at D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Database\DatabaseManager.cs:166)
Simulator.Database.DatabaseManager:Init() (at D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Database\DatabaseManager.cs:91)
Simulator.Loader:Start() (at D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Web\Loader.cs:120)
 
(Filename: D:/vmshared_space/lgsvl/simulator/Assets/Scripts/Database/DatabaseManager.cs Line: 166)

Setting up 6 worker threads for Enlighten.
  Thread -> id: 100 -> priority: 1 
  Thread -> id: 1a58 -> priority: 1 
  Thread -> id: 1634 -> priority: 1 
  Thread -> id: 2d44 -> priority: 1 
  Thread -> id: 324 -> priority: 1 
[... 171 lines cut ...]
System.Threading.ThreadPoolWorkQueue:Dispatch()
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()
 
(Filename: D:/vmshared_space/lgsvl/simulator/Assets/Scripts/Web/Modules/SimulationsModule.cs Line: 471)

Unloading 5 Unused Serialized files (Serialized files now loaded: 2)
UnloadTime: 5.985200 ms
ArgumentException: The Object you want to instantiate is null.
  at UnityEngine.Object.Instantiate[T] (T original) [0x0003d] in <f1bfefb61ab04c738a9e97a7072b0340>:0 
  at AgentManager.SpawnAgent (Simulator.AgentConfig config) [0x00001] in D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Managers\AgentManager.cs:37 
  at AgentManager.CreateAgentsFromConfigs (Simulator.AgentConfig[] agentConfigs) [0x0011c] in D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Managers\AgentManager.cs:430 
  at AgentManager.SpawnAgents (Simulator.AgentConfig[] agentConfigs) [0x00001] in D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Managers\AgentManager.cs:75 
  at SimulatorManager.Init (System.Nullable`1[T] seed) [0x002d4] in D:\vmshared_space\lgsvl\simulator\Assets\Scripts\Managers\SimulatorManager.cs:173 
[... cut ...]
  at UnityEngine.SetupCoroutine.InvokeMoveNext (System.Collections.IEnumerator enumerator, System.IntPtr returnValueAddress) [0x00027] in <f1bfefb61ab04c738a9e97a7072b0340>:0 
 
(Filename: <f1bfefb61ab04c738a9e97a7072b0340> Line: 0)

System memory in use before: 212.1 MB.
System memory in use after: 212.6 MB.

Unloading 5 unused Assets to reduce memory usage. Loaded Objects now: 5692.
Total: 5.664800 ms (FindLiveObjects: 0.440900 ms CreateObjectMapping: 0.203100 ms MarkObjects: 4.993100 ms  DeleteObjects: 0.027100 ms)

[image]

closed 2020-04-07 by EricBoiseLGSVL