lgsvl/simulator #338: About simulator5.0 XXX graph

yumianhuli2, 2019-09-02:
How to view the previous depth graph, semantic segmentation graph.. in the simulator5.0?Thanks!

martins-mozeiko, 2019-09-03:
1. Current binary release does not have this functionality.
But we will be pushing new code this week & RC release that will have functionality to see visualizations dynamically:

[image]

2. To change weather, you need to enable "interactive" checkbox for simulation. Then you'll have sliders to change these settings interactively during simulation.
Changing traffic flow will require editing C# to adjust behaviour of NPCs.
Vehicle headlights can be toggled with H button.

yumianhuli2, 2019-09-04:
Waiting for update

zelenkovsky, 2019-09-07:
Update is live. You can download that now.

closed 2019-10-02 by martins-mozeiko