lgsvl/simulator #342: BadDrawable (Invalid Pixmap or Window parameter)

chenghanzh, 2019-09-04:
I ran into this error when I started the simulator today. It did not happen yesterday. Any idea what might be going on?

[image]

martins-mozeiko, 2019-09-04:
Sometimes this happens due to issues in GPU driver. You fix it by restarting your X11 session (logout & login). If that does not help, try rebooting computer.

chenghanzh, 2019-09-05:
@martins-mozeiko I reboot the computer but still got into this problem. Should I reinstall my gpu driver?

chenghanzh, 2019-09-05:
@martins-mozeiko I reinstalled my driver and the above problem solved. However, I went into another problem. Below is my player.log file

Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 2560 x 1440 @ 60 Hz
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: count=23
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_incremental_present, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[... 8 lines cut ...]
Vulkan device type 1
Vulkan driverversion=[0x04800005] uint=[75497477]
Vulkan PSO: cache data not found [path=/tmp/LG Silicon Valley Lab/LGSVL Simulator/vulkan_pso_cache.bin]
Begin MonoManager ReloadAssembly
- Completed reload, in  0.057 seconds
WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - All passes removed
ERROR: Shader Shader is not supported on this GPU (none of subshaders/fallbacks are suitable)WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - Setting to default shader.
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
requesting resize 768 x 480
resizing window to 768 x 480
Desktop is 2560 x 1440 @ 60 Hz
InitializeOrResetSwapChain 768x480 samples=1
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
UnloadTime: 0.441253 ms
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - All passes removed

martins-mozeiko, 2019-09-05:
What is the problem you are seeing?
This log file normal. Are are any errors further down?

chenghanzh, 2019-09-05:
@martins-mozeiko The terminal says 

Set current directory to /home/disk2/lgsvlsimulator-linux64-2019.07
Found path: /home/disk2/lgsvlsimulator-linux64-2019.07/simulator
Mono path[0] = '/home/disk2/lgsvlsimulator-linux64-2019.07/simulator_Data/Managed'
Mono config path = '/home/disk2/lgsvlsimulator-linux64-2019.07/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Preloaded 'ScreenSelector.so'
Display 0 'DELL U2518D 25"': 2560x1440 (primary device).
Display 1 'K202HQL 19"': 768x1366 (secondary device).
Loading player data from /home/disk2/lgsvlsimulator-linux64-2019.07/simulator_Data/data.unity3d
Logging to /home/weizhaojin/.config/unity3d/Editor/Player.log
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 2560 x 1440 @ 60 Hz
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: count=23
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[... 25 lines cut ...]
- Completed reload, in  0.805 seconds
WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - All passes removed
ERROR: Shader Shader is not supported on this GPU (none of subshaders/fallbacks are suitable)WARNING: Shader Unsupported: 'Hidden/Nature/Terrain/Utilities' - Setting to default shader.
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
requesting resize 768 x 480
resizing window to 768 x 480
Desktop is 2560 x 1440 @ 60 Hz
InitializeOrResetSwapChain 768x480 samples=1
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
UnloadTime: 0.416646 ms
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'IndirectDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'ForwardDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - Pass 'VisibilityDXR' has no vertex shader
WARNING: Shader Unsupported: 'HDRP/Unlit' - All passes removed
simulator: Fatal IO error 0 (Success) on X server :0.
simulator: Fatal IO error 0 (Success) on X server :0.
 
and it terminates after I select the image quality and resolution

martins-mozeiko, 2019-09-05:
Last two lines sounds like some issue with X11 and vulkan driver, or how Unity is using it...
It may happen that Intel GPU does not have enough features or memory to run our simulator.

I would recommend running our simulator with Nvidia GPU, as graphics fidelity we simulate is quite high. In case you already have one in your system - that means that for some reason Unity is choosing to use Intel, not Nvidia GPU. You'll need to fix your Linux configuration to do that. One option is to use `VK_ICD_FILENAMES ` environment variable to point to nvidia icd json file.

chenghanzh, 2019-09-05:
@martins-mozeiko That works! Thank you so much!
[image]

chenghanzh, 2019-09-05:
@martins-mozeiko Sorry for bothering you, but when I click "Open Browser", it says

Mono.Data.Sqlite.SqliteException (0x80004005): Unable to open the database file
  at Mono.Data.Sqlite.SQLite3.Open (System.String strFilename, Mono.Data.Sqlite.SQLiteOpenFlagsEnum flags, System.Int32 maxPoolSize, System.Boolean usePool) [0x00077] in <79e92794a19b4995aee1b006c52ba296>:0 
  at Mono.Data.Sqlite.SqliteConnection.Open () [0x00543] in <79e92794a19b4995aee1b006c52ba296>:0 
  at Simulator.Database.DatabaseManager.Init () [0x00016] in <d7be55f55ee24966862a626c1540215e>:0 
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Cannot open database, removing it and replacing with backup
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

System.IO.FileNotFoundException: /home/weizhaojin/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/backup.db does not exist
File name: '/home/weizhaojin/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/backup.db'
  at System.IO.File.Copy (System.String sourceFileName, System.String destFileName, System.Boolean overwrite) [0x000b6] in <7d97106330684add86d080ecf65bfe69>:0 
  at Simulator.Database.DatabaseManager.Init () [0x00037] in <d7be55f55ee24966862a626c1540215e>:0 
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Cannot open backup, deleting
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

SqliteException: Unable to open the database file
  at Mono.Data.Sqlite.SQLite3.Open (System.String strFilename, Mono.Data.Sqlite.SQLiteOpenFlagsEnum flags, System.Int32 maxPoolSize, System.Boolean usePool) [0x00077] in <79e92794a19b4995aee1b006c52ba296>:0 
  at Mono.Data.Sqlite.SqliteConnection.Open () [0x00543] in <79e92794a19b4995aee1b006c52ba296>:0 
  at Simulator.Database.DatabaseManager.Init () [0x00068] in <d7be55f55ee24966862a626c1540215e>:0 
  at Simulator.Loader.Start () [0x00019] in <d7be55f55ee24966862a626c1540215e>:0 
 
(Filename: <79e92794a19b4995aee1b006c52ba296> Line: 0)

Setting up 4 worker threads for Enlighten.
  Thread -> id: 7f12f4b46700 -> priority: 1 
  Thread -> id: 7f12dffff700 -> priority: 1 
  Thread -> id: 7f12df7fe700 -> priority: 1 
  Thread -> id: 7f12deffd700 -> priority: 1 
Cannot create required material because shader is null
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Cannot create required material because shader is null
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

NullReferenceException: Object reference not set to an instance of an object
  at LoaderUI.OnStartButtonClick () [0x00005] in <d7be55f55ee24966862a626c1540215e>:0 
  at UnityEngine.Events.InvokableCall.Invoke () [0x00011] in <788595b234544a53a21c9f0449515c8f>:0 
  at UnityEngine.Events.UnityEvent.Invoke () [0x00023] in <788595b234544a53a21c9f0449515c8f>:0 
  at UnityEngine.UI.Button.Press () [0x00027] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.UI.Button.OnPointerClick (UnityEngine.EventSystems.PointerEventData eventData) [0x00011] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.EventSystems.ExecuteEvents.Execute (UnityEngine.EventSystems.IPointerClickHandler handler, UnityEngine.EventSystems.BaseEventData eventData) [0x00008] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.EventSystems.ExecuteEvents.Execute[T] (UnityEngine.GameObject target, UnityEngine.EventSystems.BaseEventData eventData, UnityEngine.EventSystems.ExecuteEvents+EventFunction`1[T1] functor) [0x00070] in <df152ef7237747c6aeec0d5524108ec4>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
UnityEngine.EventSystems.ExecuteEvents:Execute(GameObject, BaseEventData, EventFunction`1)
UnityEngine.EventSystems.StandaloneInputModule:ReleaseMouse(PointerEventData, GameObject)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMousePress(MouseButtonEventData)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent(Int32)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent()
UnityEngine.EventSystems.StandaloneInputModule:Process()
UnityEngine.EventSystems.EventSystem:Update()
 
(Filename: <d7be55f55ee24966862a626c1540215e> Line: 0)

Do you know what I can do with it? Thank you!

martins-mozeiko, 2019-09-05:
Can you check files in `/home/weizhaojin/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/` folder? Does it have `data.db` ? What are permissions on this file? This file and folder itself should be writeable by your user. In case you have run simulator with sudo it may have created this files with wrong permissions, so you'll need to chown them to your user.

You can also simply remove `data.db` file and simulator will recreate it from scratch.

chenghanzh, 2019-09-05:
@martins-mozeiko I don't see a data.db file in the folder. 

Whenever I click "Open Browser", the following message appear iteratively

NullReferenceException: Object reference not set to an instance of an object
  at LoaderUI.OnStartButtonClick () [0x00005] in <d7be55f55ee24966862a626c1540215e>:0 
  at UnityEngine.Events.InvokableCall.Invoke () [0x00011] in <788595b234544a53a21c9f0449515c8f>:0 
  at UnityEngine.Events.UnityEvent.Invoke () [0x00023] in <788595b234544a53a21c9f0449515c8f>:0 
  at UnityEngine.UI.Button.Press () [0x00027] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.UI.Button.OnPointerClick (UnityEngine.EventSystems.PointerEventData eventData) [0x00011] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.EventSystems.ExecuteEvents.Execute (UnityEngine.EventSystems.IPointerClickHandler handler, UnityEngine.EventSystems.BaseEventData eventData) [0x00008] in <df152ef7237747c6aeec0d5524108ec4>:0 
  at UnityEngine.EventSystems.ExecuteEvents.Execute[T] (UnityEngine.GameObject target, UnityEngine.EventSystems.BaseEventData eventData, UnityEngine.EventSystems.ExecuteEvents+EventFunction`1[T1] functor) [0x00070] in <df152ef7237747c6aeec0d5524108ec4>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
UnityEngine.EventSystems.ExecuteEvents:Execute(GameObject, BaseEventData, EventFunction`1)
UnityEngine.EventSystems.StandaloneInputModule:ReleaseMouse(PointerEventData, GameObject)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMousePress(MouseButtonEventData)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent(Int32)
UnityEngine.EventSystems.StandaloneInputModule:ProcessMouseEvent()
UnityEngine.EventSystems.StandaloneInputModule:Process()
UnityEngine.EventSystems.EventSystem:Update()
 
(Filename: <d7be55f55ee24966862a626c1540215e> Line: 0)

martins-mozeiko, 2019-09-05:
What about permissions on folder? Are they correct? If not either chown it, or simply delete whole folder.

chenghanzh, 2019-09-06:
@martins-mozeiko That works. Thank you so much

closed 2019-09-06 by chenghanzh