lgsvl/simulator #344: How to write C #（Script） to get the wheel angle

yangjiongfeng, 2019-09-05:
hello 
[image]
I can control the car with vehicle_cmd topic.
now,I want to know the angle of the wheel by rosbridge_server.
  How to write C #（Script） to get the wheel angle in the unity.
like the GpsDevice.cs

[image]

martins-mozeiko, 2019-09-05:
See the following comment that explains what steps you need to do publish extra data: https://github.com/lgsvl/simulator/issues/339#issuecomment-527590350
For ROS it is a bit simpler - in step 2 you should simply write C# class representing your new ROS message. For example check [Ros.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/Messages/Ros.cs) file.

yangjiongfeng, 2019-09-09:
I didn't find the Ros.cs document.
I want to create a new data type to be transmitted through rosbridge_server.

martins-mozeiko, 2019-09-09:
If you click on [Ros.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/Messages/Ros.cs) link it will open the file and you'll see the full path to it.

closed 2020-04-07 by EricBoiseLGSVL