lgsvl/simulator #345: How to control the EgoCar to follow the expected trajectory only through Python API

IsmeBlueSky, 2019-09-05:
Hi
Is it possible to control egocar to follow the expected trajectory only through Python API, like the function (NPC vehicles's follow function).  I'd like to use LGSVL to validate some simple experients，and there is no need to bridge with Apollo or Autoware, because it is troublesome for me. Thanks~

martins-mozeiko, 2019-09-05:
Currently we do not have such feature in Python API. NPCs physics is really simplistic when they are following waypoints. But we want physics of EGO vehicle to be realistic, that includes suspension, accelaration, braking, etc... Which would mean implementing whole controller & navigation logic in simulator. But that is responsibility of AD stack. This is why we expect users to connect simulator to ROS for real-time control the EGO vehicle. Python API is used only to set up the scene for specific positions and test conditions (for example, collisions).

IsmeBlueSky, 2019-09-06:
> Currently we do not have such feature in Python API. NPCs physics is really simplistic when they are following waypoints. But we want physics of EGO vehicle to be realistic, that includes [...]

 I’m grateful for your help.

IsmeBlueSky, 2019-09-07:
Vehicle steering, acceleration and braking can be controlled by keyboard. I want to use program code to control vehicle instead of keyboard control. Can you tell me where the interface for keyboard input in Python code is?Thanks！！ @martins-mozeiko

martins-mozeiko, 2019-09-09:
Python code does not deal with keyboard input. To control EGO vehicle with code you'll need to write ROS or CyberRT node and run it in Autoware or Apollo environment.

IsmeBlueSky, 2019-09-09:
> Python code does not deal with keyboard input. To control EGO vehicle with code you'll need to write ROS or CyberRT node and run it in Autoware or Apollo environment.

I still have some doubts. When the scene is loaded, I can control EgoCar's acceleration, braking and steering through the keyboard. So ,why can't I use code instead of keys :‘ →↑↓←’? This may not be a trajectory tracking, but I can always use a program to  control the car(The results may not be ideal. ). Maybe it's a boring question, hhhh.  Thanks very much!

martins-mozeiko, 2019-09-09:
In that case simulator binary is reading keyboard and controlling vehicle. Not Python.

This is done by [ManualControlSensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#manual-control). It is virtual "sensor" that processes keyboard input to control car.

If you need to control EGO car from code to follow path, you'll need write custom C# scripts. There is `apply_control` function in Python, but that is meant for very simple movements (applying constant throttle, braking, steering), not following waypoints.

closed 2020-04-07 by EricBoiseLGSVL