lgsvl/simulator #346: Attempted Lane merger - help with error messages

ninjacoder0, 2019-09-05:
Hi Team,

I got SingleLaneRoad working on Apollo + Simulator. I am now having a play to see if I can take this SingleLaneRoad, and create a junction. I found some of the explanations on your website alittle difficult to follow so could you please help.

What Ive done: 
1. I took SingleLaneRoad road, copied and rotated it to create a junction. (Image 2)
2. I copied and pasted the MapLine Boundaries as shown in the image in (image 1)
3. When running the simulation, it works as I want, I can run traffic and everything seems fine.

Problem:
Can you please help with these error messages? I find they only occur when I add Jaguar2015XE which I have not modified. 

Also can you please help me get to the stage of uploading & running on the simulator & apollo.

Big big thanks 

[image]
[image]

martins-mozeiko, 2019-09-05:
If I understand correctly you have put vehicle prefab in Unity scene directly when you are running the map?

This is not really supported, as creating vehicle requires doing extra set up work which currently is done only when vehicle is created dynamically.

What you should try is using Simulator -> Development Settings menu item. It will allow to choose which vehicle to create on the map. It will do it automatically - so there is not need to have vehicle in the scene.

ninjacoder0, 2019-09-06:
Thanks @martins-mozeiko 

I did as suggested and it worked!

When I ran the simulation I got an error message and shown in Image 1.

I then continued as following "Instructions to build standalone executable" I got to No.23 Build WebIU, and received this error message shown in image 2.

I continued to No.24 and received the a build error message shown in Image 3.

What do you suggest I do? Thanks
[image]
[image]
[image]

martins-mozeiko, 2019-09-06:
> When I ran the simulation I got an error message and shown in Image 1.

This message is normal. It is displayed to let you know that bridge cannot be connected. Either bridge is not running, or its address is incorrect.

> I then continued as following "Instructions to build standalone executable" I got to No.23 Build WebIU, and received this error message shown in image 2.

Make sure you have recent node.js version installed, and check that you have installed all required dependencies. This is done by running `npm intsall` command in WebUI folder.

ninjacoder0, 2019-09-09:
Brillant, you are correct, its worked!! Last few qu please...
I am now able to load the merge lanes in the simulator (Image 1)
Could you please help with the following:

1, I am able to "Select Vehicles" and choose Jaguar2015XE and it works without linking Apollo, although when I select Lincoln2017MKZ(5.0) in preparation for Apollo, it comes back with an invalid message, despite being built with Lincoln, above images show this? (Image 2)

2.How can I now link this simulator to Apollo. How do I create the maps needed so I can select the correct "laneMerge" dropdown (image 3). 

Again, thank you for your advice
[image]
[image]
[image]

martins-mozeiko, 2019-09-09:
1. When you say "comes back with an invalid message" - do you mean simulation fails with status "Invalid"? If so, check the logfile for more detailed information. For binary it is in `~/.config/unity3d/Editor/Player.log` location, for editor it is in `~/.config/unity3d/Unity/Editor/Editor.log` location.

2. To create Apollo map, you'll need to export its annotations (Simulator -> Export HD Map), then place generated file in `apollo/modules/map/data/MAPNAME/` folder. Then inside Apollo docker environment run `generate_map.sh MAPNAME` script (where MAPNAME is folder name under map/data folder.

ninjacoder0, 2019-09-10:
Sorry not sure what im looking for, could you guild me please.
here is the Player.log:

Desktop is 1920 x 1200 @ 60 Hz
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=14
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: count=49
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_image_format_list, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance2, enabled=0
[Vulkan init] extensions: name=VK_KHR_maintenance3, enabled=0
[Vulkan init] extensions: name=VK_KHR_multiview, enabled=0
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_relaxed_block_layout, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_sampler_ycbcr_conversion, enabled=0
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_storage_buffer_storage_class, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[... 79 lines cut ...]
  at System.Net.HttpListener.Start () [0x0000f] in <0079a30f96a047348857e1cecc6c638a>:0 
  at Nancy.Hosting.Self.NancyHost.TryStartListener () [0x00045] in <575adbde7a7c4f1da2a81d73473ed47d>:0 
  at Nancy.Hosting.Self.NancyHost.StartListener () [0x00000] in <575adbde7a7c4f1da2a81d73473ed47d>:0 
  at Nancy.Hosting.Self.NancyHost.Start () [0x00000] in <575adbde7a7c4f1da2a81d73473ed47d>:0 
  at Simulator.Loader.Start () [0x000ae] in <1e2cbf388de04c13a00510e38d56dc27>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Loader:Start()
 
(Filename: <0079a30f96a047348857e1cecc6c638a> Line: 0)

Setting up 3 worker threads for Enlighten.
  Thread -> id: 7fc831d48700 -> priority: 1 
  Thread -> id: 7fc831547700 -> priority: 1 
  Thread -> id: 7fc830d46700 -> priority: 1 
Vulkan PSO: Inconsistent pipeline cache header size[0], newDataSize[36]
GarbageCollector disposing of ComputeBuffer. Please use ComputeBuffer.Release() or .Dispose() to manually release the buffer.
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Culling group was not disposed. You have to call Dispose explicitly from the main thread. This will likely result in a crash.
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Export/Camera/CullingGroup.bindings.cpp Line: 70)

ninjacoder0, 2019-09-10:
Here is Editor.log

Desktop is 3840 x 1200 @ 60 Hz
[Package Manager] Server::Start -- Port 46769 was selected

 COMMAND LINE ARGUMENTS:
/home/artemis/Unity-2019.1.10f1/Editor/Unity
-projectpath
/home/artemis/simulator
-useHub
-hubIPC
-cloudEnvironment
production
-hubSessionId
05793a80-d3d0-11e9-87df-f5cadd66a2d0

LICENSE SYSTEM [2019910 14:54:26] Next license update check is after 2019-09-11T13:36:45

Successfully changed project path to: /home/artemis/simulator
/home/artemis/simulator
Using Asset Import Pipeline V1.
Loading GUID <-> Path mappings...0.000027 seconds
Loading Asset Database...0.045354 seconds
AssetDatabase consistency checks...0.051188 seconds
[Package Manager] Done resolving packages in 0.59s seconds
[Package Manager] Registering 39 packages:
[Package Manager]  * Package :  com.unity.inputsystem@0.2.10-preview
    (location: /home/artemis/simulator/Library/PackageCache/com.unity.inputsystem@0.2.10-preview)
[Package Manager]  * Package :  com.unity.package-manager-ui@2.1.2
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.package-manager-ui)
[Package Manager]  * Package :  com.unity.postprocessing@2.1.7
    (location: /home/artemis/simulator/Library/PackageCache/com.unity.postprocessing@2.1.7)
[Package Manager]  * Package :  com.unity.render-pipelines.high-definition@file:/home/artemis/simulator/Packages/com.unity.render-pipelines.high-definition
    (location: /home/artemis/simulator/Packages/com.unity.render-pipelines.high-definition)
[Package Manager]  * Package :  com.unity.timeline@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.timeline)
[Package Manager]  * Package :  com.unity.visualeffectgraph@5.16.1-preview
    (location: /home/artemis/simulator/Library/PackageCache/com.unity.visualeffectgraph@5.16.1-preview)
[Package Manager]  * Package :  com.unity.modules.ai@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.ai)
[Package Manager]  * Package :  com.unity.modules.animation@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.animation)
[Package Manager]  * Package :  com.unity.modules.assetbundle@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.assetbundle)
[Package Manager]  * Package :  com.unity.modules.audio@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.audio)
[Package Manager]  * Package :  com.unity.modules.cloth@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.cloth)
[Package Manager]  * Package :  com.unity.modules.director@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.director)
[Package Manager]  * Package :  com.unity.modules.imageconversion@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.imageconversion)
[Package Manager]  * Package :  com.unity.modules.imgui@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.imgui)
[Package Manager]  * Package :  com.unity.modules.jsonserialize@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.jsonserialize)
[Package Manager]  * Package :  com.unity.modules.particlesystem@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.particlesystem)
[Package Manager]  * Package :  com.unity.modules.physics@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.physics)
[Package Manager]  * Package :  com.unity.modules.physics2d@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.physics2d)
[Package Manager]  * Package :  com.unity.modules.screencapture@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.screencapture)
[Package Manager]  * Package :  com.unity.modules.terrain@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.terrain)
[Package Manager]  * Package :  com.unity.modules.terrainphysics@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.terrainphysics)
[Package Manager]  * Package :  com.unity.modules.tilemap@1.0.0
    (location: /home/artemis/Unity-2019.1.10f1/Editor/Data/Resources/PackageManager/BuiltInPackages/com.unity.modules.tilemap)
[... 541 lines cut ...]
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
Vulkan error VK_ERROR_OUT_OF_DATE_KHR (-1000001004) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
Vulkan error VK_ERROR_OUT_OF_DATE_KHR (-1000001004) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
[... cut ...]
Vulkan error VK_ERROR_OUT_OF_DATE_KHR (-1000001004) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
Vulkan error VK_ERROR_OUT_OF_DATE_KHR (-1000001004) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626
Vulkan error VK_ERROR_OUT_OF_DATE_KHR (-1000001004) file: ./Runtime/GfxDevice/vulkan/VKSwapChain.cpp, line: 626
Vulkan error./Runtime/GfxDevice/vulkan/VKSwapChain.cpp:626

martins-mozeiko, 2019-09-10:
`SocketException: Address already in use`
This error means that some other process is already listening on port 8080, so simulator cannot start. Either you are running multiple instances of simulator or there is some other process listening on 8080. You need to close other processes, or you need to change port number on which simulator is listening in config.yml file.

ninjacoder0, 2019-09-11:
Brilliant!!
Thank you all. Il close this section. You guys are truly awesome, and love your support. Again thank you.

closed 2019-09-11 by ninjacoder0