lgsvl/simulator #347: npc.getVelocity() bug or trick ?

david-gwa, 2019-09-06:
hey, lg team,

       I had a little problem with npcC.getVelocity() .     a piece of code in c# side looks like: 

```c# 
                    var npcccontroller = npc.GetComponent<NPCControllerComponent>();
                    if(npcccontroller){          
                        state.Add("velocity", npcccontroller.GetVelocity());
                        state.Add("angular_velocity", npcccontroller.GetAngularVelocity());
                        UnityEngine.Debug.Log("npc " + npc.name + "\t velocity \t" + npcccontroller.GetVelocity());  
                    }

``` 

when running python client, 

*  for some situation, it outputs: 

```script 
npc Sedan(Clone)35f9c6f5-74cd-4f41-a765-3863ca541722	 velocity 	(6.7, 0.2, 2.6)
GetVelocity() rb.vel 	 (0.0, 0.0, 0.0)  simpleVel(6.7, 0.2, 2.6)
``` 
where `simple vel` is the correct one 

* for some situations, it outputs: 

```script 
npc Sedan(Clone)2bb4e92f-cc9c-414f-9957-ec05c19d19e4	 velocity 	(0.0, 0.0, 0.0)
 GetVelocity() rb.vel 	 (5.4, 0.3, 0.0)	simpleVel	(0.0, 0.0, 0.0)
``` 
where` rigidbody velocity` is the correct one. but for this case, I actually get the (0.0, 0.0, 0.0) at Python side, as the `GetVelocity()` by default output `simpleVel` ?

is there a trick or a bug ? 

Thanks
David

david-gwa, 2019-09-06:
and this will make `npc.state.velocity` in python be zeros in some siutations

daviduhm, 2019-09-20:
Which version of simulator are you using for this? Currently, our NPCs follows simple physics with non-kinematic rigidbody, meaning that Unity just teleports the NPC object rather than smoothly moving it by applying force. Because of this, the rigidbody of NPC should not have any non-zero velocity, and the `simpleVel` is supposed to be the correct one. Could you try with the latest version of simulator and let me know which specific case did you get such result?

david-gwa, 2019-09-26:
hey, @daviduhm 

I am using  2019.04 version.   I will give an example:

**in pythonAPI I want to set npc velocity to zero:**

``` python
     s = npc.state
     s.velocity = lgsvl.Vector(0.0, 0.0, 0.0)
    npc.state = s 
```

**in Api.Commands** [AgentSetState()](https://github.com/lgsvl/simulator/blob/master-obsolete/Assets/Scripts/Api/Commands/AgentSetState.cs)

```c#
                    var rb = obj.GetComponent<Rigidbody>();
                    rb.velocity = velocity;
                    rb.angularVelocity = angular_velocity;
```

that's the problem, I expected the NPC be still, but here only rigidbody.velocity is set to 0.0, simpleVelocity is un controlled (?).  then the output looks like: 

          npc Sedan(Clone)be512087-67fb-4826-865c-8d9d5d6939b8 simple velocity: (-8.2, 0.0, 0.8) rb.velocity: (0.0, 0.0, 0.0)

and the npc is still driving in Python view

EricBoiseLGSVL, 2019-09-26:
@david-gwa 

Thanks for letting us know about this issue.  Let me try to clarify  what is going on.

When you are setting velocity using the API with AgentSetState, you are using NPC manual control.  This is not a finished feature and was a temp fix so we could work on waypoint control.  We plan on removing Manual and FixedSpeed control soon when we roll out the finished Waypoint control.

If you look at NPCController.cs in PhysicsUpdate() line 216, you can see that Manual and FixedSpeed Control never call SetTargetSpeed() that calculates simpleVelocity.  Once we release waypoint control, this will be correctly calculated.  We should have this released soon and will have much better features for direct control.

ntutangyun, 2020-02-06:
Hi, I'd like to follow up on the update.  Is it already implemented ? thanks.

EricBoiseLGSVL, 2020-02-06:
@ntutangyun we have a task currently to fix a bug with world/local velocity.  This should be released soon

ntutangyun, 2020-02-07:
@EricBoiseLGSVL thank you for your reply.

EricBoiseLGSVL, 2020-04-07:
@david-gwa @ntutangyun We are finishing up a refactor of NPC logic and should be released soon

closed 2020-04-07 by EricBoiseLGSVL