lgsvl/simulator #350: Build from source

chenghanzh, 2019-09-09:
When I build the simulator from the source, it exits before the car can run. The player.log looks like the following:

Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=72
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[... 107 lines cut ...]
  at Nancy.Hosting.Self.NancyHost.Start () [0x00000] in <575adbde7a7c4f1da2a81d73473ed47d>:0 
  at Simulator.Loader.Start () [0x00097] in <33e9e0d34f474c22a16344ddf9878a69>:0 
UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
UnityEngine.DebugLogHandler:LogException(Exception, Object)
UnityEngine.Logger:LogException(Exception, Object)
UnityEngine.Debug:LogException(Exception)
Simulator.Loader:Start()
 
(Filename: <0079a30f96a047348857e1cecc6c638a> Line: 0)

Setting up 4 worker threads for Enlighten.
  Thread -> id: 7fc45a8ae700 -> priority: 1 
  Thread -> id: 7fc45a0ad700 -> priority: 1 
  Thread -> id: 7fc4598ac700 -> priority: 1 
  Thread -> id: 7fc4590ab700 -> priority: 1 
Vulkan PSO: Inconsistent pipeline cache header size[0], newDataSize[78861]
GarbageCollector disposing of ComputeBuffer. Please use ComputeBuffer.Release() or .Dispose() to manually release the buffer.
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Culling group was not disposed. You have to call Dispose explicitly from the main thread. This will likely result in a crash.
 
(Filename: /home/builduser/buildslave/unity/build/Runtime/Export/Camera/CullingGroup.bindings.cpp Line: 70)

Do you have any idea how to fix this? Thank you!!

martins-mozeiko, 2019-09-09:
``SocketException: Address already in use```
This error message means that there is some other process already listening on port 8080 for WebUI. Either it is other simulator instance or some other program. Check who is using port 8080 and close it. Or you can change WebUI port in config.yml configuration file.

chenghanzh, 2019-09-09:
@martins-mozeiko Hi, I think I ran into another prblem. Here is my player.log again.

Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: count=72
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=0
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=0
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=0
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHR_draw_indirect_count, enabled=0
[Vulkan init] extensions: name=VK_KHR_driver_properties, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[... 132 lines cut ...]
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libuser32.dll
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libMonoSupportW.so
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libMonoSupportW.so
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libMonoSupportW.so
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libuser32.dll
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libuser32.dll.so
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/user32.dll
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libuser32
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libuser32.so
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/user32
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libuser32.dll
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libuser32.dll.so
[... cut ...]
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libMonoSupportW.so
Fallback handler could not load library /media/chenghan/disk1/lgsvl/simulator/exe/simulator_Data/Mono/libMonoSupportW.so
Receiving unhandled NULL exception
#0  0x000000008cac41 in (Unknown)
#1  0x0000000087a85b in (Unknown)
#2  0x0000000083aeaa in (Unknown)
#3  0x0000000083622a in (Unknown)
#4  0x00000000836369 in (Unknown)
#5  0x00000000bf5d00 in (Unknown)
#6  0x007f2c299206db in start_thread
#7  0x007f2c2964988f in clone

martins-mozeiko, 2019-09-09:
This kind of NULL exception usually means that there is not enough GPU memory. For BorregaAve or AutonomouStuff maps we recommend using GPU with at least 8GB of memory as these environments uses very high quality textures.

You can try closing other applications that may use GPU to see it helps. For example - Unity Editor, Visual Studio, any browser.

EsquilliDev, 2020-03-30:
@chenghanzh Did you found a fix to this issue? I'm getting exactly the same log output.

EricBoiseLGSVL, 2020-04-07:
@EsquilliDev How much GPU memory do you have and what map are you trying to load?

EsquilliDev, 2020-04-07:
So, I was able to run the simulator from source (Unity and executable) and it runs fine. The problem is running it from the standalone executable.

EricBoiseLGSVL, 2020-04-07:
@EsquilliDev Is that from a created local build or the release build?  If from the executable, do you have error logs you can post?

closed 2020-07-31 by lemketron