lgsvl/simulator #352: Web-page Formatting Issue

moenasrallah, 2019-09-10:
**Version**: 2019.09 RC1 Release

The landing page for the simulator and it's other web-pages are stuck like the image attached:

[image]

moenasrallah, 2019-09-10:
Issue resolved by clearing my browser cookies.

moenasrallah, 2019-09-10:
Reopened. Issue is consistent and always needs to have the cookies cleared. Seems to be an issue with the whole logging in process. Before, in 2019.07, it would prompt a proper re-login. With the new version, it shows the above image instead

martins-mozeiko, 2019-09-10:
Can you try pressing Ctrl+Shift+R to really reload page? This will ignore all cached files from current page (compared to just Ctrl+R refresh).

moenasrallah, 2019-09-10:
@martins-mozeiko that works as a temporary fix for now. Thank you.

closed 2020-04-07 by EricBoiseLGSVL