lgsvl/simulator #353: UI Button Causes Crash

moenasrallah, 2019-09-10:
The information "i" button on the bottom of a simulation causes the simulator to freeze/crash

Version: 2019.09 RC1

[image]

martins-mozeiko, 2019-09-10:
Can you post logfile after this freeze happen?

Depending on what you run, file is in following locations:
For Windows binary: `%APPDATA%\..\LocalLow\LG Silicon Valley Lab\LGSVL Simulator\Player.log`
For Windows Unity Editor: `%APPDATA%\..\Local\Unity\Editor\Editor.log`
For Linux binary: `~/.config/unity3d/Unity/Editor/Editor.log`
for Linux Unity Editor: `~/.config/unity3d/Editor/Player.log`

moenasrallah, 2019-09-10:
[Editor.log](https://github.com/lgsvl/simulator/files/3597392/Editor.log)

moenasrallah, 2019-09-10:
It unfroze for a quick second then refroze after a button click.

See image about "Failed to read GPU texture":
[image]

EricBoiseLGSVL, 2020-04-07:
@moenasrallah We are looking into this bug and hope to have a fix soon

EricBoiseLGSVL, 2020-04-17:
@moenasrallah we have a fix for this issue in next release

closed 2020-10-06 by EricBoiseLGSVL