lgsvl/simulator #354: Document for 2019.05 release

Kailiangdong, 2019-09-10:
Is the document of 2019.05 releases still available?. I only can find documents for 2019.07.Can some one  give me a website. Thank you.

martins-mozeiko, 2019-09-10:
Raw markdown files for old simulator are available on "master-obsolete" branch: https://github.com/lgsvl/simulator/tree/master-obsolete/Docs

It is possible to generate html format with mkdocs tool.

Note however that in a few months we will be completely removing master-obsolete branch from github to reduce size of git clone. Any development will happen only on master.

Kailiangdong, 2019-09-10:
@martins-mozeiko  Thank you very much. In addition, when I want to pull image lgsvl/apollo-3.5, I found it has same image id as lgsvl/apollo-5.0. Are they really same?

martins-mozeiko, 2019-09-10:
Yes, that is correct. Apollo 3.5/5.0 uses same base image. This image contains only Apollo dependencies. Apollo itself is built outside of the docker image.

Kailiangdong, 2019-09-10:
@martins-mozeiko Thank you I see. But when I open Dreamview. I found that the map is based on 2019.07 such as Borregas Avenue, Single Lane Road. But no san francisco

martins-mozeiko, 2019-09-10:
You should switch Apollo branch to `old-simulator`.

Kailiangdong, 2019-09-11:
@ Thank you ,it solved

closed 2019-09-11 by Kailiangdong