lgsvl/simulator #356: Material/shader for a custom map

asatria2, 2019-09-11:
Hi I'm trying to create a custom map for use with the new version of lgsvl.
I was wondering which type of material/shader should I use for the environment. The CubeTown example is using HDRP/Lit while the SingleLaneRoad is using ShaderGraph/EnvironmentSimulation. Are these for visual only, or do it have some affect on the simulation?

Thank you.

martins-mozeiko, 2019-09-11:
For most of the things you should use `HDRP/Lit` material. `EnvironmentSimulation` material should be used for roads - it is customization on Lit material to allow to apply road wetness, including puddle reflection. You can still apply `Lit` material to roads, in such case wetness setting simply won't have any effect on the road surface.

asatria2, 2019-09-12:
@martins-mozeiko  thank you for the explanation.

closed 2019-09-12 by asatria2