lgsvl/simulator #358: cannot use gpu in cpu-only caffe:check mode

Kailiangdong, 2019-09-11:
Hello. I follow the tutorial to install nvidia-docker. It is actuactlly a nvidia-container-toolkit,right.
And every things work fine.Like "docker run --gpus all nvidia/cuda:9.0-base nvidia-smi" will give nvidia infomation.
But when I build apollo in docker (./apollo.sh build_gpu). It said " # define NO_GPU LOG(FATAL)<< "Cannot use GPU in CPU-only Caffe:check mode"
[image]
What is the problem , thank you

Kailiangdong, 2019-09-11:
My docker client is 18.04, cuda 9.0 , nvidia driver 390.116, nvidia tesla p100

martins-mozeiko, 2019-09-11:
These are only compiler warnings. You can ignore those. 
Is something not working after compilation finishes?

Kailiangdong, 2019-09-11:
@martins-mozeiko .Thank you for your reply
[image]

The perception predition and planning module cannot be used. Only localization works.
I think maybe nvidia-docker problem. I didn't use nvidia-docker2,because it is deprecated in docker-ce 19.03(I am this version). So I also change code in ./dev_start.sh .
Could you tell me your version of nvidia driver and docker version? Do you use nvidia-docker2? Thank you very much.

daviduhm, 2019-09-11:
My nvidia driver is 430.30 and I'm using Docker version 19.03. What changes did you make in `docker/script/dev_start.sh`? Can you run `echo $USE_GPU` inside Apollo docker container and check if it's enabled?

daviduhm, 2019-09-12:
I'm not using nvidia-docker2 and also doing with `docker --gpus all`. Try setting `USE_GPU=1` in your dev_start.sh and make sure that you get 1 for `echo $USE_GPU` inside docker container. Otherwise, apollo will run cpu only mode and won't use your gpu.

Kailiangdong, 2019-09-17:
@daviduhm 
Hi.Execuse me. Do you change ./docker/script/dev_start.sh like this.
I return to this issue because I find my perception module works but only lidar works camera seems.
So I don't know whether gpu setting is correct and my perception module actually works? #371 
[image]