lgsvl/simulator #359: ubuntu 18.04: simulator exit with no output and no error message

zhouchengming1, 2019-09-11:
(gdb) r
Starting program: /media/psf/Home/xavier/lgsvlsimulator-linux64-2019.09-rc1/simulator 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff4cf7700 (LWP 32510)]
Found path: /media/psf/Home/xavier/lgsvlsimulator-linux64-2019.09-rc1/simulator
Mono path[0] = '/media/psf/Home/xavier/lgsvlsimulator-linux64-2019.09-rc1/simulator_Data/Managed'
Mono config path = '/media/psf/Home/xavier/lgsvlsimulator-linux64-2019.09-rc1/simulator_Data/MonoBleedingEdge/etc'
[New Thread 0x7ffff0600700 (LWP 32511)]
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Preloaded 'ScreenSelector.so'
[New Thread 0x7fffe69dd700 (LWP 32512)]
[New Thread 0x7fffe61dc700 (LWP 32513)]
[New Thread 0x7fffe59db700 (LWP 32514)]
[New Thread 0x7fffe51da700 (LWP 32515)]
[New Thread 0x7fffe49d9700 (LWP 32516)]
[New Thread 0x7fffcffff700 (LWP 32517)]
[New Thread 0x7fffcf7fe700 (LWP 32518)]
[New Thread 0x7fffceffd700 (LWP 32519)]
[New Thread 0x7fffce7fc700 (LWP 32520)]
[New Thread 0x7fffcdffb700 (LWP 32521)]
[New Thread 0x7fffcd7fa700 (LWP 32522)]
[New Thread 0x7fffccff9700 (LWP 32523)]
[New Thread 0x7fffaffff700 (LWP 32524)]
[New Thread 0x7fffaf7fe700 (LWP 32525)]
[New Thread 0x7fffaeffd700 (LWP 32526)]
[New Thread 0x7fffae7fc700 (LWP 32527)]
[New Thread 0x7fffadffb700 (LWP 32528)]
[New Thread 0x7fffad7fa700 (LWP 32529)]
Display 0 'Virtual1': 1440x900 (primary device).
Loading player data from /media/psf/Home/xavier/lgsvlsimulator-linux64-2019.09-rc1/simulator_Data/data.unity3d
Logging to /home/parallels/.config/unity3d/Editor/Player.log
[Thread 0x7fffad7fa700 (LWP 32529) exited]
[Thread 0x7fffadffb700 (LWP 32528) exited]
[Thread 0x7fffae7fc700 (LWP 32527) exited]
[Thread 0x7fffaeffd700 (LWP 32526) exited]
[Thread 0x7fffaf7fe700 (LWP 32525) exited]
[Thread 0x7fffaffff700 (LWP 32524) exited]
[Thread 0x7fffccff9700 (LWP 32523) exited]
[Thread 0x7fffcd7fa700 (LWP 32522) exited]
[Thread 0x7fffcdffb700 (LWP 32521) exited]
[Thread 0x7fffce7fc700 (LWP 32520) exited]
[Thread 0x7fffceffd700 (LWP 32519) exited]
[Thread 0x7fffcf7fe700 (LWP 32518) exited]
[Thread 0x7fffcffff700 (LWP 32517) exited]
[Thread 0x7fffe49d9700 (LWP 32516) exited]
[Thread 0x7fffe51da700 (LWP 32515) exited]
[Thread 0x7fffe59db700 (LWP 32514) exited]
[Thread 0x7fffe61dc700 (LWP 32513) exited]
[Thread 0x7fffe69dd700 (LWP 32512) exited]
[Thread 0x7ffff0600700 (LWP 32511) exited]
[Thread 0x7ffff4cf7700 (LWP 32510) exited]
[Inferior 1 (process 32506) exited with code 01]

martins-mozeiko, 2019-09-11:
Can you check error messages in `~/.config/unity3d/Editor/Player.log` file?

zhouchengming1, 2019-09-12:
No supported renderers found, exiting
 
(Filename: ./PlatformDependent/LinuxStandalone/main.cpp Line: 639)

martins-mozeiko, 2019-09-12:
What GPU do you use?

Make sure it supports vulkan and you have relevant vulkan libraries installed.
For Nvidia run `apt install libvulkan1`

closed 2019-09-15 by zhouchengming1

zhanghming, 2020-05-07:
@martins-mozeiko 
Hello, I also encounter this problem in my ubuntu 16.04 machine. My GPU is `Quadro M5000` and I have already installed the `libvulkan1`.

When I run `./simulator`, it has the following output and I can see the `Open Browser` interface but exit soon.
```shell
$ ./simulator
Set current directory to /home/z00520770/Softwares/lgsvlsimulator-linux64-2020.03
Found path: /home/z00520770/Softwares/lgsvlsimulator-linux64-2020.03/simulator
```

martins-mozeiko, 2020-05-07:
Please check the log file for more detailed error messages.
Also try running any other vulkan application like vkinfo and vkcube before running Simulator.

zhanghming, 2020-05-08:
I check the `Player.log` file, the contents are:
```shell
Mono path[0] = '/home/z00520770/Softwares/lgsvlsimulator-linux64-2020.03/simulator_Data/Managed'
Mono config path = '/home/z00520770/Softwares/lgsvlsimulator-linux64-2020.03/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 'LEN P24i-10 24"': 1920x1080 (primary device).
Loading player data from /home/z00520770/Softwares/lgsvlsimulator-linux64-2020.03/simulator_Data/data.unity3d
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
Vulkan detection: 2
Initialize engine version: 2019.3.3f1 (7ceaae5f7503)
[Subsystems] Discovering subsystems at path /home/z00520770/Softwares/lgsvlsimulator-linux64-2020.03/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHX_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: count=32
[Vulkan init] extensions: name=VK_KHR_swapchain, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_memory_requirements2, enabled=1
[Vulkan init] extensions: name=VK_KHR_maintenance1, enabled=1
[Vulkan init] extensions: name=VK_KHR_push_descriptor, enabled=0
[Vulkan init] extensions: name=VK_KHR_sampler_mirror_clamp_to_edge, enabled=1
[Vulkan init] extensions: name=VK_KHR_shader_draw_parameters, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_device_group, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_memory, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_memory_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_semaphore, enabled=0
[Vulkan init] extensions: name=VK_KHX_external_semaphore_fd, enabled=0
[Vulkan init] extensions: name=VK_KHX_multiview, enabled=0
[Vulkan init] extensions: name=VK_EXT_discard_rectangles, enabled=0
[Vulkan init] extensions: name=VK_EXT_shader_subgroup_ballot, enabled=0
[... 95 lines cut ...]
Simulator.Loader:Start()
 
(Filename: <ae22a4e8f83c41d69684ae7f557133d9> Line: 0)

Setting up 8 worker threads for Enlighten.
  Thread -> id: 7fa8537fe700 -> priority: 1 
  Thread -> id: 7fa852ffd700 -> priority: 1 
  Thread -> id: 7fa8527fc700 -> priority: 1 
  Thread -> id: 7fa851ffb700 -> priority: 1 
  Thread -> id: 7fa8517fa700 -> priority: 1 
  Thread -> id: 7fa850ff9700 -> priority: 1 
  Thread -> id: 7fa827fff700 -> priority: 1 
  Thread -> id: 7fa8277fe700 -> priority: 1 
requesting resize 1855 x 1056
resizing window to 1855 x 1056
Desktop is 1920 x 1080 @ 60 Hz
InitializeOrResetSwapChain 1855x1056 hdr=0 samples=1
Vulkan PSO: Pipeline cache has not changed skipping save handle[d568a0]
Culling group was not disposed. You have to call Dispose explicitly from the main thread. This will likely result in a crash. 
(Filename:  Line: 70)
```

It looks like the `Culling group was not disposed. You have to call Dispose explicitly from the main thread. This will likely result in a crash. ` problem, but I can't find the solution.

I installed the vulkan by using `sudo apt-get install vulkan-utils mesa-vulkan-drivers` and when I execute `vkcube`, I get `Unable to locate vkcube`. It's so strange.

martins-mozeiko, 2020-05-08:
Culling groups are not the problem, that's some internal Unity engine message we don't care about.

The problem is this: `SocketException: Address already in use`

It means that you are running other software on port 8080.  And simulator cannot use it, so it closes down. Find what other software is running, and close it down. Or change port that simulator uses in config.yml configuration file.

zhanghming, 2020-05-09:
@martins-mozeiko  Thanks a lot. Follow your suggestion, I change the port 8080 to 8800, and I can open the browser to login in the LGSVL account. However, after clicked **Login In**, the interface is stuck in `Loading...`:

[image]