lgsvl/simulator #363: About semantic camera

tecniuf, 2019-09-13:
How does the semantic camera work?
Does it use computer vision or detect objects in the hierarchy?
How can it differentiate road, cars, traffic lights, pedestrians and environment?
Could you put in a new object and identify it only by its shape?

tecniuf, 2019-09-13:
Could the camera detect those objects in the real world?

martins-mozeiko, 2019-09-13:
Our semantic camera knows what type of objects to render because simulator knows every type of object in scene. If it is road it knows to render it with road color (purple). It if is tree it knows to use tree color (green). Basically everything in map is tagged with what type of object they are. It does not do any detection on image as it already knows ground truth.

closed 2019-10-02 by martins-mozeiko