lgsvl/simulator #364: Occlusion modeling for lidar

yumianhuli2, 2019-09-15:
Are there occlusion modeling for lidar?Thank you！

martins-mozeiko, 2019-09-16:
I'm not sure what is occlusion modeling.
Current implementation of lidar is implemented through rendering depth buffer of virtual camera. Then lidar samples rendered depth at directions where lidar rays are going. Basically you can think of it as simple raytracing where each rays return closes depth (+intensity) on first hit.

yumianhuli2, 2019-09-16:
> I'm not sure what is occlusion modeling.
> Current implementation of lidar is implemented through rendering depth buffer of virtual camera. Then lidar samples rendered depth at directions where lidar [...]

Spherical ray tracker sampling mesh?

martins-mozeiko, 2019-09-16:
We do not use Unity DXR ray tracing API (yet). Lidar ray distance calculations are done from rendering depth with virtual camera.

yumianhuli2, 2019-09-17:
> We do not use Unity DXR ray tracing API (yet). Lidar ray distance calculations are done from rendering depth with virtual camera.

Have a plan for using Unity DXR ray tracing API ？Such lidar point clouds would be more realistic, allowing for true reflection and refraction.

martins-mozeiko, 2019-09-17:
Yes, we have plans to use it. Once this Unity version will be out as stable (not alpha/beta), and we'll upgrade project to it, then we'll improve the lidar sensor. Unfortunately Unity supports this feature only on Windows. It is not clear when it will be available on Linux.

EricBoiseLGSVL, 2020-04-07:
@yumianhuli2 we are currently working on this feature but Unity RTX support is not quite ready for production.  We will have more news once the feature is ready

closed 2020-04-07 by EricBoiseLGSVL