lgsvl/simulator #365: How to Established the bridge between LGSVL and Apollo5.0

Iqbalparvi, 2019-09-15:
i follow the instructions given on : 
https://github.com/lgsvl/apollo-5.0 
and run successfully  localhost:8888 server ,and run the dreamview
but cannot success to established the connection between Apollo and  simulator   after running bash Scripts/bridge.sh 
any suggestions will be appreciated
thanks

luqiang21, 2019-09-15:
@Iqbalparvi On the simulator web UI, did you choose the correct bridge when you create a simulation?

Iqbalparvi, 2019-09-16:
@luqiang21 can you please elaborate it more ,i run localhost :8080 and run the simulator as:
[image]
you can see at the bottom right corner  play button but it shows nothing like the Simulator car and hang  by just saying Running....
at the same time localhost:8888 run the dreamview 
[image]
but the car cannot move,although at the corner of this dreamview is  seems to select the same map and vehicles as in simulator end ,

martins-mozeiko, 2019-09-16:
Do not enable "Sim Control". That is internal Apollo switch to disable Cyber message processing and produce fake data for modules.

Did you enter "localhost:9090" for bridge connection when you were adding vehicle to simulation?
Can you try pressing (i) button at the bottom of simulator window when it is running - does it show that bridge is connected?

martins-mozeiko, 2019-09-16:
"Unknown operation received from client (71)" means that you have selected vehicle with ROS Bridge (for Apollo 3.0 or Autoware) and not Cyber Bridge (for Apollo 3.5/5.0). Either edit existing vehicle to choose Cyber Bridge, or choose new vehicle with Cyber Bridge for simulation.

Iqbalparvi, 2019-09-16:
@martins-mozeiko no there is no localhost:9090 in bridge list as : 
[image]
it show CyberRT.
can you let me know which vehicle support cyberRT

Iqbalparvi, 2019-09-16:
@martins-mozeiko  can you please elaborate this step:
[image]
i think i am doing some mistake here

martins-mozeiko, 2019-09-16:
localhost:8080 is for accessing web interface of simulator.

But when you add vehicle to simulation, you need to specify where to connect for sending Apollo messages. We call it bridge. You use localhost:9090 (if you are running Apollo on same computer) for bridge connection string in this window: 

[image]

This is image from https://www.lgsvlsimulator.com/docs/simulations-tab/ page

> can you let me know which vehicle support cyberRT

Any vehicle can support CyberRT bridge. Vehicle is just 3d model. What bridge is supported is just a configuration settings of "Bridge Type" and sensors json.

Make sure the simulation you are running actually has selected "Lincoln2017MKZ (Apollo 5.0)" vehicle, if that is one you want to use.

Iqbalparvi, 2019-09-16:
yes i runing apollo and simulator on the same machine (Local machine), but still unable to see and run localhost:9090

martins-mozeiko, 2019-09-16:
What do you see in bridge output terminal?
What does info button show inside simulator?

Iqbalparvi, 2019-09-16:
What do you see in bridge output terminal?
its seems like this and just stay here :
[image]
What does info button show inside simulator?
I am unable to run localhost:9090 so the simulator not show the robot car properly ......if i run localhost :8888 then it shows all the option but at the end the robot car window no pop up

martins-mozeiko, 2019-09-16:
Are you saying you cannot start simulation (play button)? It stops immediately?
If so, please check the logfile for detailed error message: https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located

Iqbalparvi, 2019-09-16:
No ,its not stop immediately  but actually there is no Robot CAR pop up in the simulation,
I find the localhost:9090 if go directly to the simulation tab like:
[image]
 the problem is that why i am unable to see the robot car and not show the bridge enviroment

martins-mozeiko, 2019-09-16:
Please check the logfile for any error messages after you start the simulation: https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located

Iqbalparvi, 2019-09-18:
@martins-mozeiko i still facing that problem ,can you please see the given screen that there is nothing automatically selected in the Enviroment and Vehicles field when i selct build option from unity ;
[image]
and also i cant see the AssetBundle folder in the simulator that i was clone ,i guess this is the reason that i still not solve the above problem .
i see other option to download the environment and vehicles manually like : 
https://content.lgsvlsimulator.com/maps/borregasave/  
this means we have create folder manually like AssetBundle inside Simulator ,confusing 
kindly guide to solve the issue 
Regards

closed 2019-09-28 by Iqbalparvi