lgsvl/simulator #366: Segmentation fault on Nvidia GTX 1050

diegoferigo, 2019-09-15:
I followed the following steps on my XPS 15 9560 with Ubuntu 18.04:

1. Installed the 2019.09 release of the simulator
1. Started the simulator with `1024x768` and `High` quality (I tried also other combinations) 
1. Created the account though the web ui
1. Downloaded the `BorregasAve` map and the `Jaguar2015XE (No Bridge)` vehicle
1. Started the simulation `BorregasAve, no bridge, data collection`

I got this output in the terminal:

```
/scratch/lgsvlsimulator-linux64-2019.09 took 3m 
❯ ./simulator
Set current directory to /scratch/lgsvlsimulator-linux64-2019.09                                                                                 
Found path: /scratch/lgsvlsimulator-linux64-2019.09/simulator
Mono path[0] = '/scratch/lgsvlsimulator-linux64-2019.09/simulator_Data/Managed'
Mono config path = '/scratch/lgsvlsimulator-linux64-2019.09/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Preloaded 'ScreenSelector.so'
Display 0 '0': 1920x1080 (primary device).
Loading player data from /scratch/lgsvlsimulator-linux64-2019.09/simulator_Data/data.unity3d
Logging to /home/dferigo/.config/unity3d/Editor/Player.log
Aborted (core dumped)
```

<details>
 <summary>Player.log</summary>

```
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.1.10f1 (f007ed779b7a)
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[... 182 lines cut ...]
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Listing simulations
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Starting simulation with id 1
 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 48)

Receiving unhandled NULL exception
#0  0x000000008cab3f in (Unknown)
#1  0x0000000087a85b in (Unknown)
#2  0x0000000083aeaa in (Unknown)
#3  0x0000000083622a in (Unknown)
#4  0x00000000836369 in (Unknown)
#5  0x00000000bf5d00 in (Unknown)
#6  0x007fb0b19d86db in start_thread
#7  0x007fb0b170188f in clone
```

</details>

As reported in https://github.com/lgsvl/simulator/issues/314 I installed all the required vulkan packages. From the log file vulkan seems fine, the problem seems to be somewhere else.

martins-mozeiko, 2019-09-16:
NULL pointer with call stack like this typically means that Unity tries to use more GPU memory that it is available.

Our BorregasAve and AutomonouStuff maps uses very high detailed textures which uses a lot of memory. It is strongly recommended to run Simulator with GPU that has at least 8GB memory. More if you are using GPU at the same time for other things - running Unity Editor, browser with a lot of content open, fancy desktop compositors, Apollo/Autoware, etc...

Try running SingleLaneRoad or CubeTown maps. If these work, then issue here is running out of GPU memory.

diegoferigo, 2019-09-21:
@martins-mozeiko Thanks for the feedback. I did some tests and that's exactly what is happening. 

[image]

It's unfortunate that so much memory is required, I have a high-end and quite recent XPS 15 laptop and at this point I cannot use it anymore for lgsvl-related activities. Is there any solution to reduce the GPU memory footprint? Choosing a very low-resolution (e.g. 640x480) does not help.

diegoferigo, 2019-09-21:
Consider that with the 2019.05 version started from a docker container with 1280x720 Ultra, and the following configuration, the GPU memory footprint is just 2250MB. The simulation is quite smooth.

[image]

Is the switch to the Web UI such demanding in terms of GPU memory?

luqiang21, 2019-09-21:
@diegoferigo It is not Web UI needs more GPU memory, it is because the new Unity HDRP and high definition textures.

diegoferigo, 2019-09-22:
@luqiang21 Is it possible / will it be possible to switch HDRP off?

luqiang21, 2019-09-22:
@diegoferigo No, the new simulator project is based on HDRP since it provides realistic lighting and high visual fidelity. As @martins-mozeiko said, you can try running SingleLaneRoad or CubeTown maps. And maybe create scenes with low-quality textures but that will decrease the visual fidelity.

diegoferigo, 2019-09-22:
Thank you @luqiang21 for the feedback. At this point I'll remain with the 2019.05 version while I figure out how to proceed. Closing.

closed 2019-09-22 by diegoferigo