lgsvl/simulator #368: How to connect Autoware and lgsvl simulator

obi56, 2019-09-16:
I want to know how to connect Autoware and lgsvl simulator.

Look at the image, you need to connect / websocket_bridge and / runlme_manager_2251_1568602169589?
[image]

Or can I just start rosbridge?

martins-mozeiko, 2019-09-16:
Please follow documentation here: https://www.lgsvlsimulator.com/docs/autoware-instructions/#launching-autoware-alongside-lgsvl-simulator

When you will select `my_sensing_simulation.launch` file and run it, it will automatically start ROS bridge node.

obi56, 2019-09-16:
There is no need to run roslaunch rosbridge_server rosbridge_websocket.launch.

Does that mean you can connect Autoware and lgsvl simulator just by clicking “Sensing”?

martins-mozeiko, 2019-09-16:
Yes. Sensing will launch all necessary modules for simulator to connect to Autoware ROS environment.

obi56, 2019-09-16:
Thank you.

Clicking "Localization" does not display the time to the right of "Ref".

And even if "Rviz" is started, it is displayed like this, and the position of the vehicle is not displayed as explained.

[image]

martins-mozeiko, 2019-09-16:
You need to click "2D Pose Estimate" button and manually specify position. This operation does not always succeed on first time. You need to repeat it until it succeeds.

obi56, 2019-09-17:
Thank you for your reply.

I won't succeed if I do it more than 10 times.

Is there any part other than 'Quick Start' in runtime manager?

Or is it because the time is not displayed to the right of "Ref" after clicking "Localization"?

EricBoiseLGSVL, 2020-04-07:
@obi5656 we are not sure why this happens but we are glad you are getting Rviz to work with Simulator

closed 2020-04-07 by EricBoiseLGSVL