lgsvl/simulator #371: Perception works but traffic light not

Kailiangdong, 2019-09-16:
Hi. 
I successfully build apollo, make it connect with lgsvl. And according to the following picture, the perception module works , it can recognize car and people. But when I set a point for ego car, it will always stop in front of traffic light. It seems that the traffic light module not works. And when I find detail in the log of traffic light. It seems that the reason is from not-working camera.
Can anyone give me the advises? Thank you in advanced.

[image]
[image]
[image]

martins-mozeiko, 2019-09-16:
What simulator and Apollo version are you using?

If vehicle does not cross intersection that usually means it does not "see" traffic light color - so it does not know if it is green. This can happen if vehicle does not have correct sensor calibration loaded.

Can you try selecting vehicle in `-- vehicle --` combobox at top right?

Kailiangdong, 2019-09-16:
@martins-mozeiko  
Thank you for your reply. I use simulator2019.05 and apollo 3.5. (Sorry , but will switch to latest simulator and 5.0 next month)
I select vehicle. Only "Mkz example" and "XE Rigged",I try both but not works. I thought I should use "XE Rigged3.5" but there is no such choice?

[image]

Kailiangdong, 2019-09-17:
I run "./perception.sh startre" . Got following log. I wonder whether problem is from is yaml files?. Because I don't find the same name of file in such forder.
[image]
[image]

Kailiangdong, 2019-09-17:
And also run the "./apollo.sh test". It seems that the unit test can be passed. But Intergration test not. 
The yolo_yolo_obstacle_detector_test not work. I thought maybe camera's problem?
**Thank you in advanced for someone's help.**
(My host is 18.04, nvidia tesla p100, compute capability 6.0, driver 390.116, I use docker lgsvl/apollo3.5 and simulator 2019.05 release.
[image]
[image]

EricBoiseLGSVL, 2020-04-07:
@Kailiangdong please update to latest simulator and let us know if you are still having issues

closed 2020-04-07 by EricBoiseLGSVL

Martin20, 2020-04-16:
I updated and still have the problem.

EricBoiseLGSVL, 2020-04-16:
@Martin20 This issue is closed.  Please create a new issue so we can be sure to assign someone