lgsvl/simulator #372: Do you have the documention of the old version 2019-05?

gaohongfein, 2019-09-18:


martins-mozeiko, 2019-09-18:
Old simulator is available in branch: https://github.com/lgsvl/simulator/tree/master-obsolete/Docs/docs

Note that will be removing it because we are not developing it anymore. This is to save size of git clone.

gaohongfein, 2019-09-19:
Thanks, and more question, how to annotate the speed bump in old version?

gaohongfein, 2019-09-19:
I use the temp waypoint to create the speedbump, when i export the hdmap, it occures these problem.

KeyNotFoundException: The given key was not present in the dictionary.
System.Collections.Generic.Dictionary`2[TKey,TValue].get_Item (TKey key) (at <f2e6809acb14476a81f399aeb800f8f2>:0)
Map.Apollo.HDMapTool.MakeSpeedBumpAnnotation () (at Assets/Scripts/Map/HDMapTool.cs:1502)
Map.Apollo.HDMapTool.Calculate () (at Assets/Scripts/Map/HDMapTool.cs:964)
Map.Apollo.HDMapTool.ExportHDMap () (at Assets/Scripts/Map/HDMapTool.cs:122)
HDMapToolEditor.OnInspectorGUI () (at Assets/Scripts/Editor/HDMapToolEditor.cs:23)
UnityEditor.InspectorWindow.DrawEditor (UnityEditor.Editor[] editors, System.Int32 editorIndex, System.Boolean rebuildOptimizedGUIBlock, System.Boolean& showImportedObjectBarNext, UnityEngine.Rect& importedObjectBarRect) (at /home/builduser/buildslave/unity/build/Editor/Mono/Inspector/InspectorWindow.cs:1374)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

martins-mozeiko, 2019-09-19:
We are not supporting old simulator code anymore. Annotation and hd map export code has greatly improved and is changed significantly in new version. I suggest you to upgrade to new version.

Francisobiagwu, 2019-09-27:
I prefer the old version to the new version. It seems to have exactly what I need; I am in the process of modifying some of the source files to suit my needs. Its so unfortunate that it is no longer supported.

closed 2020-04-07 by EricBoiseLGSVL