lgsvl/simulator #373: Launching Autoware alongside LGSVL Simulator

obi56, 2019-09-18:
Self-position estimation is not performed.
I want to execute like this sample image.
[image]

In the 'Quick Start' tab, the time is not displayed on the right side of "Ref" of "Localization" like the image.
Is there a solution?
[image]

Is there anything else to set other than the 'Quick Start' tab?

Please help me.

obi56, 2019-09-18:
The reason why self-location estimation cannot be executed is that ROSBAG data is not clicked on "Play" in the "Simulation" tab?

Where is ROSBAG data distributed for autoware simulation on lgsvl simulator?

hadiTab, 2019-09-19:
@obi56 there is no need for a rosbag. Rosbags are used for replaying previously collected data (which is what the autoware quickstart is using). Since the simulator provides a live stream of data, no rosbag is used.

There could be several reasons to why the localization is not working. We will need more information to figure out why. Here are several things you can do:

- After launching localization, look at the terminal that Autoware was launched from and see if there are any error messages. If so please share them with us.

- Make sure that you have downloaded the maps properly. We use git-lfs to store the pointcloud map in the autoware-data repository. If you do not have git-lfs installed the map will not be downloaded properly. A simple check will be to check the size of the PCD file for the map located in: `~/shared_dir/autoware-data/BorregasAve/data/map/pointcloud_map/BorregasAve.pcd`
If it is incredibly small (less than 1KB) that would mean that the file has not been downloaded and you should install git-lfs and download it.

obi56, 2019-09-20:
When "Map" tab is pressed, "OK" is displayed, but this error appears.
[image]

Pressing "Localization" tab does not output an error.
[image]

I think the download using git-lfs has been successful.
[image]

hadiTab, 2019-09-20:
@obi56 seems like the map is there. I suspect that the localization is actually searching for a solution. It usually takes a while. Those numbers appear after it finds a pose. To help it out you can open up rviz and set an initial pose on the map near where the vehicle is located (by default this would be at the bottom of the map you see in rviz).

obi56, 2019-09-30:
Even if you run '2D Pose Estimate' many times, the time is not displayed in the 'Localization' tab.

Is this an error related?
[image]

hadiTab, 2019-10-07:
Can you run `rosnode info /ndt_matching` and paste the results here?

obi5656, 2019-10-07:
Thank you.

[image]

hadiTab, 2019-10-08:
I don't see anything that seems out of place. Unfortunately, I don't really have a solution for you.
In general, the ndt_matching localization is a bit finicky and it's usually the hardest part to get going correctly. I usually wait for a few minutes after enabling localization, and then set the 2D pose and it almost always works. If it doesn't I'll restart everything and try again. It's a bit painful, but allowing it to sit for a few minutes seems to somehow help it with searching for a correct pose. The folks at Autoware may have a better solution for this.

If you don't really care about using lidar-based localization and just want some form of localization working, another option is to use the GPS sensor. This is also not straight forward because of an issue with the GPS implementation in Autoware that makes it only work with Japanese coordinates. There is a fix for this (these three pull requests: [1](https://gitlab.com/autowarefoundation/autoware.ai/core_perception/merge_requests/26), [2](https://gitlab.com/autowarefoundation/autoware.ai/common/merge_requests/20), [3](https://gitlab.com/autowarefoundation/autoware.ai/utilities/merge_requests/27)). You'll need to apply these changes to the autoware source code and either run autoware from source or rebuild a docker image with the modified source code to run with the simulator. Once this is done you can use the `gnss_localizer` node and set it to plane 0 and it should provide good position estimates. It is a bit of extra work.

obi5656, 2019-10-08:
Thank you.
I'll give it a try.

closed 2020-04-07 by EricBoiseLGSVL